arendst/Tasmota #4: Telemetry with Uptime only

evilgreen, 2017-01-29:
TELEMETRY messages sent periodically are used to report various parameters in a single JSON.
Sometime, probably when the new hour of operation begins, TELEMETRY message is sent with only Uptime in the body.

The problem is: I'm using openHAB to extract a single parameter from TELEMETRY message by using jsonpath transformation. When message with Uptime only arrives this generates an error as there is no parameter of interest is present.
`tele/sonoff1/TELEMETRY {"Time":"2017-01-29T19:59:17", "Uptime":2, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"crazy", "RSSI":48}}
tele/sonoff1/TELEMETRY {"Time":"2017-01-29T20:02:30", "Uptime":3}
tele/sonoff1/TELEMETRY {"Time":"2017-01-29T20:04:23", "Uptime":3, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"crazy", "RSSI":44}}
`
May I suggest to either send complete messages all the time or move Uptime to a separate topic?
Thanks!

arendst, 2017-01-30:
Can't you do some testing on the JSON fields you receive and want to act upon? In case of a Sonoff Pow you'll get two TELEMETRY messages with different fields so that would also be a problem in your case.

```
09:27:27 MQTT: tele/pow1/TELEMETRY = {"Time":"2017-01-30T09:27:27", "Uptime":17, "POWER":"OFF", "Wifi":{"AP":2, "SSID":"indebuurt2", "RSSI":100}}
09:27:27 MQTT: tele/pow1/TELEMETRY = {"Time":"2017-01-30T09:27:27", "Energy":{"Yesterday":"0.096", "Today":"0.000", "Period":0, "Power":0, "Factor":"0.00", "Voltage":0, "Current":"0.000"}}
```

KPK75, 2017-01-30:
I see the same issue in logs:

_tele/sonoff1/TELEMETRY = {"Time":"2017-01-30T10:02:30", "Uptime":1}
_tele/sonoff2/TELEMETRY = {"Time":"2017-01-30T10:02:30", "Uptime":1}
_tele/sonoff1/TELEMETRY = {"Time":"2017-01-30T10:12:11", "Uptime":1, "POWER":"OF...
_tele/sonoff2/TELEMETRY = {"Time":"2017-01-30T10:13:27", "Uptime":1, "POWER":"OF...
 
Uptime is sent on every hour at 2:30 and full status at TelePeriod time

arendst, 2017-01-30:
As designed. Even if teleperiod is 0 you'll receive the uptime message every hour as a kind of heartbeat to be used as trace information.

klond, 2017-01-30:
This build will only support MQTT JSON messages in the future?

evilgreen, 2017-01-30:
Additional filtering on the message content is absolutely possible. What I'm trying to promote is a "fixed data set per item" approach. As a 2nd option I want to mention what is already in place for RESULT topic, where "Info1" and "Info2" are used to distinguish between 2 data sets.
I believe this will provide simpler and easier to parse output. Thanks to @arendst for all the efforts!

arendst, 2017-01-30:
@klond Yes, this build will only support MQTT JSON except for the power on/off which will also send messages like ```stat/sonoff/power ON``` to be able to use optional MQTT Retain.

@evilgreen The advantage of JSON is that a fixed dataset is not needed as all messages are labelled with their unique label. The telemetry data will change depending on what sensors you connect to your sonoff.

klond, 2017-01-30:
@arendst  Thanks for the information, I understand that JSON have alot of advantages .
I just need to find out how to convert/filter this in OpenHAB. 
I could just imagine a OpenHAB binding around this :)

evilgreen, 2017-01-30:
@arendst  I see your point, but I still believe it's a bad idea to put everything in one basket. Feel free to close the issue.

@klond In opehHAB you will use the regular MQTT binding, configuration is straight forward. Please feel free to ask questions in OH community.

ThomDietrich, 2017-01-31:
@klond https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB

klond, 2017-01-31:
@ThomDietrich  Nice job, I was thinking the same.. If we could make some examples for each Sonoff product.  Then it would be easier to deploy.

ThomDietrich, 2017-01-31:
@klond yes that would be really handy! If you'd like, join me in filling the wiki article as well as the community forum thread. I will add Sonoff Basic, Dual and Pow (hopefully) this evening.

ThomDietrich, 2017-02-02:
@arendst I've encountered the "problem" you describe in your first answer. The first TELEMETRY message doesn't contain Power, the second doesn't contain RSSI. I'm parsing for both, seeing the same issues @evilgreen was seeing.

```
2017-02-02 23:16:14.756 [WARN ] [b.core.events.EventPublisherDelegate] - given new state is NULL, couldn't post update for 'WZ_Expedit_Power'
2017-02-02 23:16:14.794 [WARN ] [b.core.events.EventPublisherDelegate] - given new state is NULL, couldn't post update for 'WZ_Expedit_RSSI'
```

Now, Yes I agree json is great, openHAB is easily able to parse it and further information for new sensors or similar can be added over time. I believe we all agree there.
The problem is, that Sonoff-Tasmota sends different datasets in parallel on one MQTT topic. That seems error prone (as proven by openHAB, see above) and isn't really meaningful, because one of the big advantages of MQTT is, that you can define as many topics as you wish. You should seriously consider fixing the set of fields per topic. Topics could be "Status", "Uptime", "Energy", "Telemetry", ...

arendst, 2017-02-04:
Version 3.9.11 makes a difference between hourly update and teleperiod information.

ThomDietrich, 2017-02-04:
Thanks @arendst!
As always, great work ;) I've tested the new firmware with my Sonoff Pow and can confirm everything to work as expected. I'll update the wiki article on openHAB right away.

I think this issue can be closed but I can't speak for @evilgreen.

evilgreen, 2017-02-04:
Thanks @arendst for listening to his fans ;)
Yes, this issue can be closed as all seems to be more structured than before.
@ThomDietrich appreciate your contribution!

PaulHam211, 2017-02-04:
I came on here to post the same issue, saw that it had been fixed, downloaded the new bin and uploaded over the air! Perfect. FIXED! 👍 
Thank you.

closed 2017-02-04 by arendst