arendst/Tasmota #5: 4ch goes into reboot loop if gpio7 gpio7 are configured for ds18b20

aequitas, 2017-01-29:
using: v3.9.5
device: Sonoff ch4 v1.1

Setting GPIO7 or GPIO8 to `4` (DS18x20) results in the device rebooting. There is serial and MQTT output. Website is not accessible. 

Other modes not experimented, GPIO2 does not exhibit this issue so my problem is resolved.

arendst, 2017-01-30:
Thnx for the feedback.

That behaviour was kind of expected as pins GPIO07 and GPIO08 are being used to connect the flash chip to the ESP8285. As the flash interface is using DOUT mode these pins are the data pins being used.

For the moment I will remove them from the configuration until I have the 4CH myself and can do some more testing but I'm afraid they will be gone forever.

closed 2017-01-30 by arendst

aequitas, 2017-01-30:
When looking at the PCB layout pin 7 and 8 seem to be routed towards an unpopulated space for a 2.5 jack connector. Much like the TH10. There are two jumpers disconnecting the jack connector from the ESP. I think there was some intent for expansion here. GPIO7 and GPIO8 are SPI according to the datasheet. But without CS (2.5 jack only has 4 lines) I don't think that is the intended type.

arendst, 2017-01-30:
You're right. But the pins are also used by the flash on the ESP8285 die so that's probably why they refrained from using the 2.5 jack...

aequitas, 2017-01-30:
Maybe the flash could be disabled after the program is initially loaded into memory to f ree up those pins or disabled during IP with the peripheral?

If you need a 4ch to test I can miss mine for a couple of days, did I guess correct you also live in the Arnhem region?

arendst, 2017-01-30:
As there is more of us I live in the Groningen region ;-)

The flash is physically connected to these pins. Every time you switch your relay the state is also written to flash to preserve it's last state at next power on. In fact all settings are stored in flash and some of them are updated when you change them.

Perhaps some wizzkid can make it work but I do not see a bright future ahead with this option...

aequitas, 2017-01-30:
That's a bit to much of a travel for me :). 

If I ever need the pins or get bored I will try to work out a solution. I don't know yet if the flash has a CS connected to the ESP but if it does it should be implementable by adding proper locking for sensor read and flash writes.