arendst/Tasmota #6: Error, missing file  #include <NeoPixelBus.h>

ozeebrown, 2017-01-30:
Hi,
I'm testing your latest code on sonoff devices. I'm getting the following error when verifying the code.

C:\Users\GuyPERERA\Documents\Arduino\Sonoff-Tasmota-master\sonoff\xdrv_ws2812.ino:31:25: fatal error: NeoPixelBus.h: No such file or directory
 #include <NeoPixelBus.h>

Can you please give me a possible solution. Thanks.

arendst, 2017-01-30:
You'll have to install the NeoPixelBus library using the IDE Library Manager.

ozeebrown, 2017-01-30:
That's what was missing. Thanks you.
But, a different issue. WHen I test the buttons. Only No.1 relay is working. Other 3 buttons are not.

sonoff 4CH is working fine on v3.1.3

I guess, I'll wait till you get your 4CH.

ThomDietrich, 2017-01-30:
Another solution: Use Platform.IO, which will install libraries and set correct settings on it's own. https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite

closed 2017-01-30 by arendst