arendst/Tasmota #7: WS2812b LEDs with unusal GRB mapping

phedders, 2017-01-30:
Is there an easy way to configure an LED string that is mapped weirdly - where Red and Green are swapped?

I was wondering about adding another LedTable.

Apologies if I missed the obvious. I'm on 3.9.7 IIRC.
And again many thanks for an awesome fw!

arendst, 2017-01-30:
As a test change the colortype in xdrv_ws2812.ino from
```
NeoGrbFeature
```
into
```
NeoRgbFeature
```
and see what happens.

phedders, 2017-01-30:
What a pain.

I changed line 34 - and tried that - no change - but then I'm not using pin3

I cannot change 36 because (I think) that feature is not compatible with the bitbang lib (I've hit that before on my code)

/home/prh/Sonoff-Tasmota/sonoff/xdrv_ws2812.ino: In function 'void ws2812_init()':
/home/prh/Sonoff-Tasmota/sonoff/xdrv_ws2812.ino:462:9: error: cannot convert 'NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBangMethodBase<NeoEsp8266BitBangSpeed800Kbps> >*' to 'NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBangMethodBase<NeoEsp8266BitBangSpeed800Kbps> >*' in assignment

I'll try using DMA (pin 3/rx)

arendst, 2017-01-30:
You have to change lines 34, 36 , 460 and 462!

phedders, 2017-01-30:
Tried on GPIO03 - and still RG are swapped:

FTR I also set in user_config.h:

 129 #define USE_WS2812                               // WS2812 Led string support (+8k code, +1k mem)
 130   #define USE_WS2812_DMA                         // DMA supports only GPIO03 (= Serial TXD) (+1k mem)

Thanks Theo.

phedders, 2017-01-30:
OK I'm an idiot - changed all of those occurrences and it does work just fine with RG swapped 👍 
Thanks for your patience. Do you think this could be configurable?

arendst, 2017-01-30:
Fixed in version 3.9.10 just released

closed 2017-01-30 by arendst

phedders, 2017-01-30:
You're amazing - Thanks for such a quick response. Would it be possible to configure on a running system? I have a mix of 2812's and of course most are GRB - just a few RGB (the traditional LED shape ones) and it would be nice to have one firmware image for them all.

If that is not possible I totally understand :)

arendst, 2017-01-30:
Looked at it and it's just too much overhead as a new object needs to be defined https://github.com/Makuna/NeoPixelBus/issues/103

phedders, 2017-01-30:
Oh thats a shame - thanks again.