arendst/Tasmota #9: button behavior

evilgreen, 2017-01-30:
I want to use (actually, to keep) the button behavior described in wiki:

> 1 short press: Toggles the relay either directly or by sending a MQTT message

In fact I had this working before but now with v3.9.10 it does not work for me anymore.
I have a separate buttontopic (sonbutt1), so I want Sonoff to send a message only and not toggle the relay - this will be handled externally.

This is what I see on the broker when the button is pressed (there is no external logic in place):

> cmnd/sonbutt1/POWER TOGGLE
> cmnd/sonoff1/power ON
> stat/sonoff1/RESULT {"POWER":"ON"}
> stat/sonoff1/POWER ON

Why "cmnd/sonoff1/power ON" is here? Please advise how to restore the desired button behavior. Thanks!

arendst, 2017-01-30:
In what version did it work as expected?

You're right that the first line is wrong already; should have been ```cmnd/sonbutt1/POWER ON``` or OFF and no TOGGLE

evilgreen, 2017-01-31:
Cannot be specific about version, but it was first configured in December using the current at this time build. Then it was working until I switched from the standard to this new build. Not sure, though.
In the log: "APP: Multipress 1", sleep is 0.
Thanks!

arendst, 2017-01-31:
Just tried it myself. No problems detected. It just only executes ```cmnd/sonbutt1/POWER TOGGLE```.

You must have a MQTT service somewhere reacting on this command and executing your ```cmnd/sonoff1/POWER ON```

What happens if you press the button again? Do you see a command like ```cmnd/sonoff1/POWER OFF```?

evilgreen, 2017-01-31:
Sorry, that was my fault - it was my previous OH installation on another server "reacting on this command and executing your cmnd/sonoff1/POWER ON".
Thank you for checking!

closed 2017-01-31 by evilgreen