arendst/Tasmota #12: Button Header on Dual

hahmann, 2017-02-01:
Not sure exactly if it is feature request or a fix but on the Dual I would expect the button header (GND/Button0/Button1/GND) to work with external wall switches like on the basic (GND/GPIO14), it is not working for me, button1+gnd does nothing, button0+gnd reset to defaults.

Doing some research, realized that the protocol on the serial for the F330 is the same in both ways, which means that enabling the relays or receiving a switch/button change from the F330 is the same as the protocol you have already implement, the difference is that it has to read serial and enable the relays accordingly.

I've tried to understand the code and make my own changes but my coding skills are limited.

By the way, amazing project!
Regards,
Gabriel.

arendst, 2017-02-11:
Duals button header is not supported

closed 2017-02-13 by arendst

xxpigeonkillerxx, 2017-06-05:
I am having this same issue. 

It would be great for the button header to be controlled.

This would allow the header pin labeled Button0 and the pin labeled Button1 to be set up and function the way a Sonoff Basic functions when you enable GPIO14 to operate as a "Switch". 

This would allow the Dual to operate in conjunction with 2 mechanical wall switches and really unlock the power of the Dual.

@arendst Is this feature possible to develop? Or does the way the hardware is setup with the ESP having to go thru the F330 chip to control the relay, make it impossible to implement "Switch" functionality for the header pins Button0 and Button1?

xxpigeonkillerxx, 2017-06-05:
I flashed the software onto my Dual and have configured it through the front end to operate as "Sonoff Dual" as opposed to "Sonoff Basic". 

When I simulate a button press by tying the Button0 pin to ground, and then releasing the ground connection, one of the relay toggles state. When I do the same with Button1 the other relay toggles.

However, this toggle is not recognized by the software in anyway. The state does not toggle on the front end. MQTT messages are not fired. 

I see that these header pins (Button0 and Button1) are tied directly to the F330 chip. 

Is it even possible to make the ESP module aware of what is happening on the Button0 or Button1 pins?