arendst/Tasmota #13: Any chance of HC-SR501 PIR motion detector support?

mihalski, 2017-02-01:
Although specified as requiring at least 5V it is possible to run them at 3.3V without any problems as per http://techgurka.blogspot.com.au/2013/05/cheap-pyroelectric-infrared-pir-motion.html

Publishing state changes via MQTT would allow for all sorts of interesting automation.

Regards,
Michal

Hogesyx, 2017-02-02:
Hi Michal, as SR501 behave more or less like a wall switch(aka high when triggered), you can actually just use the standard wall switch method.

You can also split the power and trigger using a different button topic.

mihalski, 2017-02-02:
Thanks!

The first pet I figured, but I was not aware I could plait power and trigger. I'll have a look at that when I get home.

I really wish there was an additional platform supported (ie Generic 8266), which let me have access to all the I/O as this glorious firmware works perfectly well on any module but within the limitations of the Sonoff I/O's.

mihalski, 2017-02-02:
Unfortunately I cannot test this for at least a week as I'd need to take a working Sonoff and try to mod it and I only get one day off work a week.

Currently using an Adafruit Feather Huzzah for prototyping the functionality/automation and it just so happens that it doesn't break out GPIO#3 so I'm all out off luck after using 5 and 6 for I2C.

https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/pinouts

KPK75, 2017-02-02:
Yes, it is possible. Just edit last position of mytmplt modules in sonoff_template.h.

mihalski, 2017-02-02:
I thought this would be where it would be done, but it's not clear to me what the format of these entries is.

Is it documented somewhere?

KPK75, 2017-02-02:
It's easy to figure it out : 17 positions from GPIO0 to GPIO16, on each position you define what it does : KEY, LED, RELAY etc.

mihalski, 2017-02-02:
And the 0's are what? Not connected?

KPK75, 2017-02-02:
Or not used.

don-willingham, 2017-02-03:
The HC-SR501's I ordered arrived today. I hope to try hooking one up to an ESP-01 this weekend. I'll share my results.

mihalski, 2017-02-03:
I've added the following to the bottom but it's complaining about too many initialisers:

```
  { "Huzzah",          // Adafruit Feather Huzzah
     GPIO_USER,        // GPIO00 Red Led (use as output, and blink the red LED)
     0,
     GPIO_USER,        // GPIO02 Blue Led (use as output, and blink the red LED)
     0,
     GPIO_USER,        // GPIO04 Optional sensor
     GPIO_USER,        // GPIO05 Optional sensor
     0, 0, 0, 0, 0, 0,
     GPIO_USER,        // GPIO12 Optional sensor
     GPIO_USER,        // GPIO13 Optional sensor
     GPIO_USER,        // GPIO14 Optional sensor
     GPIO_USER,        // GPIO15 (use as output)
     GPIO_USER         // GPIO16 Optional sensor
  }
```

Not seeing where I'm messing up.

arendst, 2017-02-03:
Hint: Perhaps you need to change something on line 58 too...

BTW I do not support so many GPIO_USER pins as the sensor code is no-entrant. YMMV.

mihalski, 2017-02-03:
DOH!! Thank you! I totally missed that.. Compiling fine now.

I don't see myself using more than I2C plus a motion sensor at this stage. It's more a matter of choosing the GPIo's they go on than anything.

Looking forward to trying this out.

mihalski, 2017-02-03:
Well up until now things were working fine, but are no longer working fine..

I updated via USB.. perhaps the cause of the problems? And after updating it started to use the default project wifi access point settings.. So I changed (both) to my access point and re-uploaded.. STILL trying to use the old ones..

Not sure WTF I've done but hopefully I can figure it out. Any tips are welcome.

KPK75, 2017-02-03:
For a good start just use GPIO_KEY1, GPIO_KEY2, GPIO_LED1, GPIO_LED2 etc and leave other as GPIO_NONE and see if it works..

mihalski, 2017-02-03:
It doesn't matter what I do.. It's permanently stuck on the old default access points and then commits suicide. Used to work fine for me too but after adding it to the module types it's not working anymore and even if I revert to the original code the problem persists.

```
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-2609...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP1 indebuurt1 in mode 11N as sonoff-2609...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute
00:00:33 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

arendst, 2017-02-03:
Hint 1: CFG_HOLDER
Hint 2: commands ssid1 and password1
Hint 3: Take a itsy bitsy tiny view at the wiki. May I suggest the commands tab...

mihalski, 2017-02-03:
Thank you! CFG_HOLDER was the key.

I'd never needed to use it before so it seems it sort of became invisible to me.

Still falls over if I try to switch to my Huzzah template so I will try KPK75's suggestion to simplify the number of GPIO's I make available.

mihalski, 2017-02-04:
I can't get my defined Huzzah module to work. Whenever I choose it It crashes the firmware and then restarts with Sonoff Basic preselected. As an example here's a log of Me choosing User Test Module, which works fine, and then Huzzah which crashes and resets to Sonoff Basic:

```
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-2609...
00:00:04 RSLT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:00:04 Config: Use default configuration
00:00:05 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-2609...
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-2609.local with IP address 192.168.1.77
00:00:08 MQTT: Attempting connection...
00:00:09 MQTT: Connected
00:00:09 MQTT: tele/sonoff/LWT = Online (retained)
00:00:09 MQTT: tele/sonoff/RESULT = {"Info1":{"Module":"Sonoff Basic", "Version":"3.9.10", "FallbackTopic":"DVES_C6AA31", "GroupTopic":"sonoffs"}}
00:00:09 MQTT: tele/sonoff/RESULT = {"Info2":{"WebserverMode":"Admin", "Hostname":"sonoff-2609", "IPaddress":"192.168.1.77"}}
00:00:10 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:10 MQTT: stat/sonoff/POWER = OFF
00:00:17 MQTT: tele/sonoff/TELEMETRY = {"Time":"1970-01-01T00:00:17", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":76}}
18:25:58 HTTP: User Test Module, GPIO2 0, GPIO3 0, GPIO4 0, GPIO14 0
18:25:59 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-2609...
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-2609.local with IP address 192.168.1.77
00:00:08 MQTT: Attempting connection...
00:00:09 MQTT: Connected
00:00:09 MQTT: tele/sonoff/LWT = Online (retained)
00:00:09 MQTT: tele/sonoff/RESULT = {"Info1":{"Module":"User Test", "Version":"3.9.10", "FallbackTopic":"DVES_C6AA31", "GroupTopic":"sonoffs"}}
00:00:09 MQTT: tele/sonoff/RESULT = {"Info2":{"WebserverMode":"Admin", "Hostname":"sonoff-2609", "IPaddress":"192.168.1.77"}}
00:00:10 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:10 MQTT: stat/sonoff/POWER = OFF
18:26:17 MQTT: tele/sonoff/TELEMETRY = {"Time":"2017-02-04T18:26:17", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":82}}
18:26:21 HTTP: Huzzah Module, GPIO4 0, GPIO5 0, GPIO13 0, GPIO14 0
18:26:23 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-2609...
00:00:04 RSLT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:00:04 Config: Use default configuration
00:00:05 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_C6AA31, GroupTopic sonoffs) Version 3.9.10
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-2609...
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-2609.local with IP address 192.168.1.77
00:00:08 MQTT: Attempting connection...
00:00:08 MQTT: Connected
00:00:08 MQTT: tele/sonoff/LWT = Online (retained)
00:00:08 MQTT: tele/sonoff/RESULT = {"Info1":{"Module":"Sonoff Basic", "Version":"3.9.10", "FallbackTopic":"DVES_C6AA31", "GroupTopic":"sonoffs"}}
00:00:08 MQTT: tele/sonoff/RESULT = {"Info2":{"WebserverMode":"Admin", "Hostname":"sonoff-2609", "IPaddress":"192.168.1.77"}}
00:00:09 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:09 MQTT: stat/sonoff/POWER = OFF
18:26:47 MQTT: tele/sonoff/TELEMETRY = {"Time":"2017-02-04T18:26:47", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":80}}
```

And here's my definition of Huzzah:
```
  { "Huzzah",          // Adafruit Feather Huzzah
     GPIO_LED1,        // GPIO00 Red Led (use as output, and blink the red LED)
     0,
     GPIO_LED2,        // GPIO02 Blue Led (use as output, and blink the red LED)
     0,
     GPIO_USER,        // GPIO04 Optional sensor
     GPIO_USER,        // GPIO05 Optional sensor
     0, 0, 0, 0, 0, 0,
     GPIO_NONE,        // GPIO12 Optional sensor
     GPIO_SWT1,        // GPIO13 Optional sensor
     GPIO_USER,        // GPIO14 Optional sensor
     GPIO_NONE         // GPIO15 (use as output)
  }
```
I'm certain that there must be something I am overlooking.

arendst, 2017-02-04:
I can reproduce and will investigate......

mihalski, 2017-02-05:
You sir are a champion of the people!

It all works perfectly :)

And splitting up the STATE and SENSOR is very helpful too.

closed 2017-02-05 by arendst