arendst/Tasmota #15: Topic name with dash/underscore

ThomDietrich, 2017-02-01:
When the MQTT topic is set to e.g. "sonoff-E8A6E4" the topic actually published is "sonoff_E8A6E4".

Why is that?

Edit: Could this be related to the fact that "MQTT Client and Fallback Topic" and "MQTT Topic" are the same on my modules? It's actually nowhere documented what the difference between the two of them is.

arendst, 2017-02-02:
Well the fall back topic is equal to the MQTT Client as the client has to be unique for MQTT to work. Using this feature every sonoff can still be addressed even if you have (accidently) set two sonoffs to the same Topic.

I never thought of the fact where someone would use the same name for both fall back topic AND Topic. I do not recommend it as it makes unique identification difficult.

If you first set MQTTClient and then want to set Topic using the same name the software does not allow it and sets Topic to the default as in user_config.h. I do not check the other way round and that's probably how you did it.

A name with a dash as in ``sonoff-1234`` as Topic is perfectly fine. Same for MQTTClient.

ThomDietrich, 2017-02-02:
Understood. That's more or less as I expected it. Thanks for explaining

In that case I would like to continue with the suggestion to default to a unique name for the Topic. Defaulting to "sonoff" doesn't make sense if you own more than one module - which is most likely for 99% of Sonoff users...

Additionally I did not understand your decision to work with two id schemes, the last six MAC digits and a four digit decimal I can not relate to. Might be beneficial to use the MAC scheme only, wdyt?

arendst, 2017-02-03:
Will look into topic uniqueness

arendst, 2017-02-04:
No need for topic uniqeueness as the fall-back topic is already unique.

The standard topic is set to ``sonoff`` so that when one new sonoff is flashed the user can configure it right away accessing it by the topic ``cmnd/sonoff/..``. The first thing to do than is renaming it to a **user known** unique name.

The use of the four digits decimal is a consequence of keeping unique hostnames as short as 15 character to satisfy Windows WINS Netbios naming.

ThomDietrich, 2017-02-04:
> configure it right away accessing it by the topic cmnd/sonoff/... The first thing to do than is renaming it to a user known unique name.

For that purpose one can subscribe for `cmd/+`. That's how MQTT is designed... http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices

How would one deal with multiple freshly flashed modules? They would all receive the same new settings. If a user does the configuration via http (as I do) the default topic doesn't matter, it's just a nuisance to change it... In both cases a unique topic to begin with would have been easier.

Don't get me wrong, you know that I'm a fond user of your firmware! The process is simply a bit unclear. I'm not sure how a less experienced user will deal with these things. Intuitive usability is sometimes hard to get right. If "The first thing to do than is renaming it to a user known unique name." is part of your intended process, would be good to at least document it somewhere.

arendst, 2017-02-04:
using cmnd/+ addresses all devices in your environment and not just a few sonoffs (I have many different models like tickers, lcds, lights all not using this software). This does NOT address a unique device. So a grouptopic to address same kind of firmware is just a need.

The user selectable unique topic is a pet name used for all outgoing communication and is therefore mandatory.

Dealing with multiple new installs is done using the fallback topic so no problem at all.

Finally this software is to be used by someone willing to invest some time in getting to know a little about MQTT. All other features are making it nice to have.

ThomDietrich, 2017-02-05:
I agree that we disagree 😃  
However I think we both made our point and what you are saying is totally valid of course.
I will try to add a few details to the wiki.

closed 2017-02-05 by ThomDietrich

ThomDietrich, 2017-02-06:
@arendst great to see you've added a lot of sensors content. I did create the following menu entry/article for that: https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Modification

Just so you know. You might want to delete the article, rename the Initital-Configuration article or move the content. As you prefer.

RyanEwen, 2018-04-16:
Nevermind.. I see that this is supported in the development branch according to https://github.com/arendst/Sonoff-Tasmota/issues/2409