arendst/Tasmota #16: Sonoff LED

jaysuk, 2017-02-02:
I don't seem to be able to adjust the brightness of the LED's.
Currently, they are not as bright as a sonoff LED with the default firmware

arendst, 2017-02-03:
Still waiting for mine to arrive...

ecsfang, 2017-02-04:
Hi, just received my Sonoff LED. Not tested to much yet (loaded with Theo's latest), but power on/off works, as well as Dimmer 0-100. But I have noticed that the leds blink shortly (quick power off on) when I change the dimmer value.
I also had a problem with the led-bars, half of them was wrongly mounted, there is a plastic cover with arrow indication of polarity, but in half of mine, the cover was backwards, so make a check and check the indication on the actual flex (the are arrows in both ends there as well).
Cheers,
Thomas

ecsfang, 2017-02-05:
Hi again!

Did some more simple testing after uploading 3.9.11. The "Dimmer" command only works for the "cold" led, in my case, the "warm" led is always at about 50%, but both can be turned on or off. The led-strip contains two different leds, which has its own output, one "cold" white and one "warm" white.
Nor is the code stable yet: every now and then the leds blinks shortly (while being powered on).
So, I guess that we just have to wait for Theo's devices to arrive :)

Best regards,
Thomas

arendst, 2017-02-05:
Does a dimmer2 command do anything to the warm led?

ecsfang, 2017-02-05:
Hi! Yes,  Dimmer2 handles the warm led, thanks -  I missed that one! :) 
But still, when power on - i.e. Leds are turned on,  the leds randomly powers off and on.

arendst, 2017-02-18:
For those interested: I received my Sonoff Led and fixed the low dimmer values; it appears the default analogWrite range is from 0 to 1023. I fixed that so the full range of light is available. Currently working on:
- one color command for both leds like a normal RGB led strip
- one dimmer command for both leds like a normal RGB led strip

I hate to mention that the flickering is indeed caused by the background runs so that could be a spoiler.

To be continued...

ecsfang, 2017-02-18:
Sounds great - thanks for your findings!
I have not yet seen a schematic of this device, but I will try to find out if any GPIOs are available for e.g. an external switch.

ecsfang, 2017-02-18:
A quick look shows that GPIO 4, 5 and 15 are easily accessible (reserved for RGB?).  So with some small modifications these should be available for other use. I will try to add an external push-button switch to be able to turn the leds on/off with a smooth transition. I already have the wirings for an old lamp in the kitchen which I want to replace, but still use the physical switch to turn them on/off.

closed 2017-03-08 by arendst