arendst/Tasmota #17: Keeps reconnecting to AccessPoint

yaydogan, 2017-02-02:
I am having a strange issue, not sure if anyone else runs into it. 

I am not using MQTT and enabled Wemo simulation. I can connect to the Wifi Manager, configure SSID/Password. The device will restart and connect to access point. When I monitor the logs on serial port, I see device is connecting to access point periodically. Looks like it is loosing wifi connection and connecting again. And when I connect to webserver to see/change config, I often get time outs due to wifi connection issues. Anyone seeing similar issues? 

Thanks in advance.

yaydogan, 2017-02-03:
Did more testing and have additional observations. When I have MQTT_USE set to 1 but MQTT server is not existing, the switch becomes unresponsive at times and the wifi connection also keeps dropping. I compiled with MQTT_USE set to 0 and all works perfectly.  I currently don't have a way to test with MQTT_USE = 1 and a real MQTT server.  

Could be the unsuccessful MQTT connection being retried periodically causing the issues.

don-willingham, 2017-02-03:
Try pointing MQTT to an IP address that exists, even it's not running MQTT.  If it doesn't exist, it'll wait for some timeout period before it gives up.  If it fails fast, it may be more responsive. I'm running https://github.com/arendst/Sonoff-MQTT-OTA-Arduino, and until I disabled MQTT, Alexa was slow to control my sonoff.

arendst, 2017-02-03:
If you do not need MQTT why not simply disable it on the fly using the Configuration webpage?

Stop fooling around with the user_config.h page. I'll make it as small as possible.

yaydogan, 2017-02-03:
Closing it. Disabling it will work just fine.

closed 2017-02-03 by yaydogan