arendst/Tasmota #19: Service: http hosting binary images

smadds, 2017-02-03:
Not sure if this is useful for anyone, but I am hosting the binary image on an http server to allow Sonoffs to update via URL (Github insists on https which is beyond the standard configuration of firmware).

Every hour, on the hour I get the latest binary from 
> https://github.com/arendst/Sonoff-Tasmota/raw/master/api/arduino/sonoff.ino.bin
and host it at
> http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin

I was wary of doing this with the old version of the software, due to the number of tweaks needed in the config file before compiling. Still, if you want the raw mqtt-ota version from
> https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/raw/master/api/arduino/sonoff.ino.bin
it's also available at
> http://sonoff.maddox.co.uk/mqtt-ota/sonoff.ino.bin

wiktorschmidt, 2017-02-04:
@smadds maybe you could provide an alternative approach (using Github releases). I described it here:
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/pull/260

smadds, 2017-02-04:
Looking at your suggestion again, @wiktorschmidt, I read it as making a secure connection to create a local copy on a trusted machine on the local network. Is this correct?

ThomDietrich, 2017-02-05:
@smadds that's a nice gesture of yours. I went ahead and included your URL in an automatic upgrade example over in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB
I hope that is fine with you. Thanks!

I'll also subscribe to this thread to stay up to date with changes.

wiktorschmidt, 2017-02-05:
@smadds sorry I wasn't clear

What I meant is that instead of linking to
https://github.com/arendst/Sonoff-Tasmota/raw/master/api/arduino/sonoff.ino.bin
you can link to the automatically compiled (via TravisCI) firmware.bin file that can be found here:
https://github.com/arendst/Sonoff-Tasmota/releases/latest
is a little bit more tricky as there is no "direct link" to the bin file - you need to extract it from JSON via github API: https://api.github.com/repos/arendst/Sonoff-Tasmota/releases/latest it's `assets[0].browser_download_url`

Thanks for putting it together!

ThomDietrich, 2017-02-05:
@wiktorschmidt Interesting! Regarding the openHAB solution above. I will stay with the direct link for now. JSON parsing is an easy job in openHAB but would definitely make the example less comprehensible.

wiktorschmidt, 2017-02-05:
@ThomDietrich absolutely - this would have to be implemented by @smadds to make sense. Obviously parsing JSON is not a good idea on the openHAB side :)

One can also look at https://github.com/carolynvs/github-release-proxy - this solves the http/https problem quite elegantly.

ThomDietrich, 2017-02-05:
At the end of the day it would probably be best if @arendst hosted an "official" build

wiktorschmidt, 2017-02-05:
@ThomDietrich it IS hosted by @arendst on github. The only problem is that it's only available via https

ThomDietrich, 2017-02-05:
@wiktorschmidt no, @smadds is providing a local copy. At least that's what he says in his first posting ;)

smadds, 2017-02-05:
Not sure about local, @ThomDeitrich, my HTTP copies are hosted on a cloud VPS.

Yes, happy for anyone to access them as a contribution to this excellent project.

Oh the power of issuing an mqtt "cmnd/sonoffs/update" !

smadds, 2017-02-11:
Just to say that I have extended the number of .bin files available by using platformio on the server to locally compile the Tasmota branch for different boards. The script now looks every 10 minutes for a change in version number and recompiles the targets if required.

I have added nodemcu and ESP8285 (sonoff touch etc.) to the standard sonoff versions. The URLS are:-

- http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin
- http://sonoff.maddox.co.uk/tasmota/sonoff-nodemcu.ino.bin
- http://sonoff.maddox.co.uk/tasmota/sonoff-8285.ino.bin

If anyone needs another target that platformio is capable of compiling on Linux then let me know. (RPi is NOT currently available as it needs compiling on an RPi)

The old branch is still being copied from the github repository hourly as before.

arendst, 2017-02-11:
Starting with version 3.9.12 there is no need to compile specifically for the ESP8285 if the image is uploaded by webpage or OTA.

If the device is configured for Sonoff 4CH OR Sonoff Touch, both ESP8285,  the downloaded ESP8266 firmware image is updated with the specific ESP8285 DOUT option before it gets written to the device flash during reboot.

For serial uploaded images it is still MANDATORY to use the DOUT flag during compiling and uploading.

jaysuk, 2017-02-11:
Do I still need to compile separate images for devices with 1M memory and ones with 4M memory?

arendst, 2017-02-11:
Well if you want to...
I use 1M images on all my sonoff related hardware so also on ESP12Es and the Pow which has 4M too.

ecsfang, 2017-02-22:
Hm, @smadds, looking at your hosted link, the binary is the 3.9.16 sw, but TASMOTA is already up in 3.6.19 ... ?
Has your script stopped running?

Cheers!
Thomas

smadds, 2017-02-23:
My apologies, I was replacing the platformio script with one that compiled to more targets, and that ran in to a dependency problem when new features were added. Theo has confirmed that the other versions are no longer needed, so I have dropped this.

From now the http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin is up to date (and should keep up to date!), but I have dropped all of the other versions (including the legacy version).

ecsfang, 2017-02-23:
No problem! I was just confused when I updated and realized that I had actually downgraded to 3.9.16 instead of upgrading ...

Thanks for the service! ;)

ThomDietrich, 2017-02-23:
@smadds Upgrade article updated accordingly. Thanks!
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade

smadds, 2017-02-27:
I have added an MQTT server to this VPS. It's pretty locked down (no publishing accepted) and has just one topic to subscribe to: `sonoff-version`. The URL is `sonoff.maddox.co.uk` using the standard port `1883` with no authentication needed.

If you have mosquitto-clients installed, the following should return the version I am serving:
`mosquitto_sub -h sonoff.maddox.co.uk -t 'sonoff-version'`

Hoping this will be useful to trigger auto updates for some users.

ThomDietrich, 2017-04-05:
Hey @smadds, would you be willing to host additional builds for the available variations?
I've deployed a Sonoff Basic with multiple DS18B20, which needs a firmware with the [USE_DS18x20](https://github.com/arendst/Sonoff-Tasmota/blob/master/sonoff/user_config.h#L132) directive enabled.

smadds, 2017-04-06:
Hi @ThomDietrich, yes - happy to. The only question is how to get them there.

If Theo @arendst could add variations to his Platformio script, that would ensure the variations are updated with new releases (I now collect & host everything from the /api/arduino folder). I realise this would make the release time longer, and there could be a lot of different requests.

Alternatively I could compile different versions on my server, but I don't know the compiler well enough (at the moment!) to know how to pass selective compilation directives.

CommodoreWhite, 2017-04-20:
Hi. I've followed the link in the wiki and followed the build method as described by quickly which uses atom atom/platformio. For the life me i cant figure out how to use it to create my own firmware.bin file.

Am i right that you guys use the arduino ide to create the firmware.bin file, if so, how.? I can't​ use the stock firmware.bin file because my dragons stubbornly refuse to honour the poweronstate setting so i need to tinker with the code and create my own firmware.bin to update with using OTA. c.f. issue #284 

I'd thought platformio should be able to do the job but at present I can only make it program my dragons using serial.

Regards

smadds, 2017-04-20:
If platformio allows cross-compilation under linux then I could do it, @CommodoreWhite . I say that because the RPi does not currently cross compile, and platformio needs a host RPi board to compile on.

CommodoreWhite, 2017-04-20:
Really don't understand what RaspberryPie has got to do with anything. I always do software development on Windows. Are you sure you're responding to the right message?

Regards, Confused of Suffolk

⁣Sent from BlueMail ​

davidelang, 2017-04-20:
platformio creates firmware.bin files, arduino IDE creates sonoff.ino.bin

CommodoreWhite, 2017-04-20:
Aha! During a compile of tasmota i saw firmware.bin being created but never found it once the compile had completed so i assumed it was created as an intermediate file and then promptly deleted. Any chance you could give me a clue where platformio leaves it? Thanks.

⁣Sent from BlueMail ​

davidelang, 2017-04-20:
no idea, I haven't tried using it yet.

On Thu, 20 Apr 2017, CommodoreWhite wrote:

> Aha! During a compile of tasmota i saw firmware.bin being created but never found it once the compile had completed so i assumed it was created as an intermediate file and then promptly deleted. Any chance you could give me a clue where platformio leaves it? Thanks.
k

CommodoreWhite, 2017-04-20:
> No, just misread your msg, @CommodoreWhite
> <https://github.com/CommodoreWhite>. Thought quickpi was a variant of the
> RPi board.

Ok. Quickpi is a guy who created a YouTube video showing how to compile and
upload tasmota using platformio and is recommended on the tasmota wiki.

Any clue where platformio stashes the firmware.bin file?

CommodoreWhite, 2017-04-20:
Sorry, messages crossed in the post thanks for your timely responses anyway.

Peter

smadds, 2017-04-20:
.pioenvs/sonoff/firmware.bin

smadds, 2017-06-02:
hi Theo @arendst

What do you mean by "Removed pre-compiled versions from repository as they are available within the release" in release 5.1.4 ? Are they somewhere else in Github, or do you mean that they can be compiled locally?

It's a shame not to have the "vanilla" build, as I was using those to copy for distribution via http.

arendst, 2017-06-02:
They are somewhere else on Github!

Look via the releases tab at the top of the page: https://github.com/arendst/Sonoff-Tasmota/releases

smadds, 2017-06-02:
Yes, I found them. Just got to figure a way of automatically copying them from there. Was much easier copying *.bin from a cloned github folder

smadds, 2017-06-02:
OK, I have updated my script to copy over the 4 releases based on the release version number. Should cope with future version changes, but won't spot any extra builds. If I'm missing one, please let me know!

Apologies to anyone who tried to use the .bin versions of 5.14 over the last few hours from my server - the MQTT flag will have updated but not the binaries. Please re-flash now.

ThomDietrich, 2017-06-02:
Not sure what you are doing but you should always be able to retrieve the latest release and binaries linked to those via the github json responses. A simple example can be seen here: https://github.com/yann25/Sonoff-Tasmota_autoflash/blob/master/Sonoff-Tasmota_autoflash/tasmota_autoflash.py#L34

smadds, 2017-06-02:
Thanks, @ThomDietrich. I've updated my bash script with the api command. Was fairly simple - although at the moment has not got the error checking your Python script has:
`curl 'https://api.github.com/repos/arendst/Sonoff-Tasmota/releases' | jq -r '.[0] .assets[] .browser_download_url' | xargs wget -N`

The folder http://sonoff.maddox.co.uk/tasmota/ will now have a copy of the all assets from the latest release by the time my MQTT topic is updated.

ThomDietrich, 2017-06-02:
Wonderful.
Btw. I've kept the wiki in sync with the important links from this thread. Would you like to continue doing so? https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade

smadds, 2017-06-02:
Wiki updated

nburrows72, 2018-02-28:
I appreciate this is an old thread, and the following is slightly off topic.
As this is just an observation, didn't want to raise a new issue.

I came across this thread while looking how to get notification about tasmota releases.
I subscribed via MQTT to sonoff.maddox.co.uk/sonoff-version, but this is sending out the current version every 5 minutes.  (Checked via MQTT-spy).  

FYI: I also found out you can get a RSS feed from github of the tasmota releases using:
https://github.com/arendst/Sonoff-Tasmota/releases.atom

Thanks @smadds for host the binaries via http / mqtt notification mechanism.

smadds, 2018-02-28:
You're right, @nburrows72 ! Looks like my script logic is faulty and I'm comparing the release ID with the latest commit id - no matter how many times I pull the latest release they won't match after an interim update has been made.

Anyone with better git & shell experience than me like to suggest a better way to compare local and remote release IDs? This is my current code:

 ```
cd /git-source-location
echo $(date)
git remote update
latestTag=$(git describe --tags `git rev-list --tags --max-count=1`)
if [ $(git rev-parse HEAD) != $(git rev-parse @{u}) ]
then    echo "Updating now..."
        echo "From version:" $latestTag
        cd /binaries-location
        curl 'https://api.github.com/repos/arendst/Sonoff-Tasmota/releases' | j$
        cp sonoff.bin sonoff.ino.bin -f
        mosquitto_pub -t "sonoff-version" -m $latestTag -r -u xxxxxxxxxxx -P xxxxxxxxxx
        echo "Updated"
        echo "New version:" $latestTag
else
        echo "Already up to date - " $latestTag
fi
```

If not I'll have a look when I get a minute...

JueBag, 2018-05-01:
High @smadds 
I did use the Tasmota. bin file provided by your server for updating my Sonoff devices in an OPENHAB installation. Thanks for that.
The change to 5.13 however does not work automatically, although it should. The incooperated OtaMagic (needed now because of the increased size) needs the minimal image for the two step process. My guess is that this file is missing on your server, but as said only a personell guess!
Is there a chance to check that and place the minimal image on the server as well?
If needed I could test the process from here.

smadds, 2018-05-01:
Hi @JueBag 

I tried upgrading some devices to 5.13 last night and the normal OTA did not work for me either.

My script copies all the release files from Github, and sonoff-minimal is definitely there (http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin)

I found I had to upload the minimal image first then do the ota to the normal sonoff.bin. I have just tried ota from 5.13 to 5.13.1 and that failed too.

Theo - is there something we can change to automate this process (e.g. can we store the rom type so that minimal does not overwrite it and so a new version can be automatically fetched after the minimal has loaded?)

arendst, 2018-05-02:
Same is true for upgrading from 5.12.0 to 5.13.1.

```
00:00:00 Project sonoff Wemos4 (Topic wemos4, Fallback DVES_83BB10, GroupTopic sonoffs) Version 5.12.0
00:00:00 CFG: Saved to flash at FA, Count 39, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as wemos4...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on wemos4.local with IP address 192.168.2.223
00:00:06 APP: (UTC) Wed May 02 07:47:29 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
09:47:30 MQT: Attempting connection...
09:47:30 MQT: Connected
09:47:30 MQT: tele/wemos4/LWT = Online (retained)
09:47:30 MQT: cmnd/wemos4/POWER = 
09:47:30 MQT: Subscribe to cmnd/wemos4/#
09:47:30 MQT: Subscribe to cmnd/sonoffs/#
09:47:30 MQT: Subscribe to cmnd/DVES_83BB10/#
09:47:30 MQT: tele/wemos4/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_83BB10","GroupTopic":"sonoffs"}
09:47:30 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
09:47:30 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
09:47:30 MQT: homeassistant/switch/wemos4/config =  (retained)
09:47:31 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
09:47:31 MQT: stat/wemos4/POWER = OFF
09:47:38 MQT: tele/wemos4/STATE = {"Time":"2018-05-02T09:47:38","Uptime":"0T00:00:15","Vcc":2.726,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":92,"APMac":"00:22:6B:FE:8E:20"}}
09:47:45 HTP: Main Menu
09:47:54 HTP: Firmware Upgrade
status 4
09:48:04 CMD: status 4
09:48:04 RSL: Group 0, Index 1, Command STATUS, Data 4
09:48:04 MQT: stat/wemos4/STATUS4 = {"StatusMEM":{"ProgramSize":487,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
09:48:10 HTP: Upgrade started
09:48:10 RSL: Group 0, Index 1, Command OTAURL, Data http://sonoff.maddox.co.uk/tasmota/sonoff.bin
09:48:10 MQT: stat/wemos4/RESULT = {"OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
09:48:10 RSL: Group 0, Index 1, Command UPGRADE, Data 1
09:48:10 MQT: stat/wemos4/RESULT = {"Upgrade":"Version 5.12.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
09:48:10 CFG: Saved to flash at FB, Count 40, Bytes 1648
09:48:13 HTP: Web server stopped
09:48:13 UPL: http://sonoff.maddox.co.uk/tasmota/sonoff.bin
09:48:13 HTP: Web server active on wemos4.local with IP address 192.168.2.223
09:48:15 HTP: Web server stopped
09:48:15 UPL: http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin
09:48:37 HTP: Web server active on wemos4.local with IP address 192.168.2.223
09:48:38 MQT: Attempting connection...
09:48:38 MQT: Connected
09:48:38 MQT: tele/wemos4/LWT = Online (retained)
09:48:38 MQT: cmnd/wemos4/POWER = 
09:48:38 MQT: Subscribe to cmnd/wemos4/#
09:48:38 MQT: Subscribe to cmnd/sonoffs/#
09:48:38 MQT: Subscribe to cmnd/DVES_83BB10/#
09:48:39 MQT: stat/wemos4/UPGRADE = Successful. Restarting
09:48:40 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc9f0c112
@cp:0
ld

00:00:00 APP: Boot Count 15
00:00:00 Project sonoff Wemos4 (Topic wemos4, Fallback DVES_83BB10, GroupTopic sonoffs) Version 5.13.1-minimal-2_4_0
00:00:00 WARNING This version does not support persistent settings
[... 53 lines cut ...]
00:00:04 HTP: Web server active on wemos4.local with IP address 192.168.2.223
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/wemos4/LWT = Online (retained)
00:00:06 MQT: cmnd/wemos4/POWER = 
00:00:06 MQT: Subscribe to cmnd/wemos4/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/DVES_83BB10/#
00:00:06 MQT: tele/wemos4/INFO1 = {"Module":"Sonoff Basic","Version":"5.13.1","FallbackTopic":"DVES_83BB10","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
00:00:06 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
00:00:07 MQT: stat/wemos4/POWER = OFF
00:00:14 MQT: tele/wemos4/STATE = {"Time":"1970-01-01T00:00:14","Uptime":"0T00:00:14","Vcc":2.973,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":90,"APMac":"00:22:6B:FE:8E:20"}}
00:00:28 MQT: Attempting connection...
00:00:28 MQT: Connected
00:00:28 MQT: tele/wemos4/LWT = Online (retained)
00:00:28 MQT: cmnd/wemos4/POWER = 
00:00:28 MQT: Subscribe to cmnd/wemos4/#
00:00:28 MQT: Subscribe to cmnd/sonoffs/#
00:00:28 MQT: Subscribe to cmnd/DVES_83BB10/#
00:01:03 APP: (UTC) Wed May 02 07:50:45 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
```

NOTE: As OtaMagic is introduced first in release 5.12.0 this only works starting from release 5.12.0 as it uses code introduced in 5.12.0. Previous versions can only be upgraded by manual OTA uploading the minimal version first before uploading the final version.

smadds, 2018-05-02:
Just checked with serial monitor and indeed it worked fine from 5.13.0 to 5.13.1. 

Can I suggest an update to the onscreen text? "Device will restart in a few seconds" is now a bit optimistic! Suggest something like:

Device upgrade in progress. This involves uploading and installing 2 firmware updates, so please be patient - it could take 2-3 minutes.

ascillato2, 2018-08-25:
All this information is added to the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade#firmware-binary-sources

Thanks a lot :+1:

closed 2018-08-25 by ascillato2