arendst/Tasmota #20: PWM control on any pin

phedders, 2017-02-03:
I've been looking at GPIO_PWM* and I'm not sure that they can be used as general PWM control of an exernal device - I think they are just for the SONOFF LED?

(I want to dim 12v/30W LED fixtures via external MOSFETS using PWM)

Do you have code coming/ready that could do this? If not I'll have a go hacking something in tonight.

arendst, 2017-02-03:
No I won't support it. Too much deviations. Try ESP-Easy.

phedders, 2017-02-03:
Thanks - I'll do that or hack in my own. I love the rest of your firmware... which is why I want to do everything in it :)

hoekbrwr, 2017-02-12:
Are you really making software for IoT? Then PWM is a must! Otherwise it is useless as IoT. I thought Sonoff is about Home Automation. You make RGB with WS2812 and not having something simple as PWM for LED lighting? Very strange! It perfectly fits into the Sonoff SV module. I think it is even possible to get it in the housing completely with PowerFETs(and take out relay for space if necessary). Maybe this is more a problem Itead should solve. They are now selling driver for sort of array of RGB LEDs and not for 1-color LEDs.

phedders, 2017-02-15:
+hoekbrwr I really do not think demanding like that is helpful. I do wish things were different and I think this is would be a great opportunity missed, and if provided with a great pull request Theo might be persuaded to change his mind... You aren't a paying customer with a mandate to complain. Theo also kindly pointed to an alternative that way well work for me (and you) but which I have not yet tested.

You and I also have the opportunity to maintain our own forks thanks to the nature of OS and Theos generosity.

hoekbrwr, 2017-02-15:
I am not demanding anything! I am only stunned by the omission of the feature. Sorry, but I am not a user who can easily fit it in a fork of the code! So I just hope the solution will come from either Itead or Theo.
Btw I have tested the dimmerx commands on a SV configured as a LED unit. Unfortunately it does not work well. I got the 50% problem you mentioned too. Als random flashing when powered up or when reset by OTA updates, so no solution through the backdoor. So I have to use homemade units with separate ESP-12 and Pete Scargill's firmware, not with a Sonoff unit

altelch, 2017-02-15:
ESP8266 has only software PWM. There are known issues with low ans high duty cycles (flickering) especially when you have network traffic. I think there won't be a satisfying solution using software PWM if your device has still some other work to do. What I could think of is the use of hardware pwm with somethin like https://www.adafruit.com/product/815. I2C is impemented so we would "just" need a driver and maybe a board layout like my IRremote extension to fit into the sonoff. Currently I don't have a need for this feature so I won't look into this...but maybe someone else can write the code as that's how this project works. 

Greetings,
   Heiko.

closed 2017-03-08 by arendst