arendst/Tasmota #22: Sonoff RF issue

mverzakov, 2017-02-04:
Hi!
I'm struggling with a Sonoff RF Problem.
I'm trying to pair my RF button with sonoff rf. (I don't havany special connection for R21 )
So I pushed button on RF module, LED blinked red one time. I bushed my RF button and LED blinked 3 times.
Now, every time I push my RF button, Sonoff's Led blinks red, but power doesn't switch.
Could you please help me?

MacWyznawca, 2017-02-21:
I have the same problem. Sonoff RF learned remote control. Pressing the remote control, causing LED light red, but does not switch. I set Sonoff RF module in configuration.

mverzakov, 2017-02-22:
I was somehow able to control Sonoff RF with a remote. 
I'm not sure what was the problem.

davidelang, 2017-03-18:
I'mm close this issue since you seem to have it working, you can re-open it if you still have problems.

closed 2017-03-18 by davidelang