arendst/Tasmota #23: Request - BME280 support

smadds, 2017-02-04:
I know space is tight, but I'm hoping to standardise on the [BME280](http://static.cactus.io/docs/sensors/barometric/bme280/BST-BME280_DS001-10.pdf) temperature/humidity/pressure sensor. Is there any possibility of adding I2C support for it in Tasmota?

([Adafruit](https://github.com/adafruit/Adafruit_BME280_Library) seems to be the most popular library, or this [cactus.io](http://cactus.io/hookups/sensors/barometric/bme280/hookup-arduino-to-bme280-barometric-pressure-sensor) one more stand-alone)

jaysuk, 2017-02-04:
The BME280 is already supported

smadds, 2017-02-04:
OK thanks @jaysuk. I was mistakenly looking for it in the list of sensor options. 
Do you know how to access it? Does allocating SCL and SDA pins enable a search for sensors?

jaysuk, 2017-02-04:
yes, that should be sufficient.

arendst, 2017-02-04:
For some I2C devices power cycling of sonoff is needed.

closed 2017-02-04 by arendst

marksmanaz, 2020-07-17:
I am compiling Tasmota to add support for the seven segment display and I want to use the BME280 sensor but it wont load drivers. I have tried #define USE_BME280 in the user override but it won't load. i2cscan finds the sensor. can I include the driver when compiling on my own? I can't use the sensor bin because it doesn't include my display driver.

Thanks