arendst/Tasmota #24: Some information only published via JSON on boot?

fiskfan, 2017-02-04:
Again. thank you for your excellent work. you should really fix a donate button...

Question. it appears as if some telemetry information is only published on boot? I'm specifically thinking on the /RESULT topic which contains Info1 and Info2 messages with Version, IP Address and device type information in the JSON message. - Is there a way to have the device publish it at a interval similar to how it publishes Uptime and SSID for example?

The reason why i ask is that since the IP address, version and device type MQTT messages doesn't get stored then i loose this information when my MQTT broker goes down and after that i need to reboot all of my sonoff devices in order to get the info back.

Thanks again.

arendst, 2017-02-04:
What's wrong wit just executing ```status 0``` at regular times and receive a lot of info?

fiskfan, 2017-02-04:
Well.. that would work just fine i suppose. i was not aware about the command an how much info it did output it did send. I will scheduelle or send this command manually when needed.

thanks

closed 2017-02-04 by arendst