arendst/Tasmota #28: Request: Legacy mode for MQTT messages

tam481, 2017-02-05:
Hi Theo,
Is this possible in this version or is it too much work? It makes life a lot easier for those using OpenHAB2.

arendst, 2017-02-06:
No, legacy mode messages had to go due to memory restraints

Mickbaer, 2017-02-07:
Hi Theo,
first many thanks for that great work.
As you have changed the MQTT message mode from legacy to Json,
the results, to be send to my FHEM are very different.
I'm not so firm to programming FHEM to get the received string  to be used at an variable or plot file.
I want to use the values send with this string 
ENERGY 
{"Time":"2017-02-06T02:14:52", "Yesterday":"0.424", "Today":"0.145", "Period":0, "Power":0, "Factor":"0.00", "Voltage":227, "Current":"0.000"} 
but I do not know how to do.
So my question is, if you can post an example how to work with the new results format.

Thanks for your help in advanced.
Greatings from Mickbaer (Berlin Germany)

arendst, 2017-02-13:
I can't help you with FHEM but there must be something on their forum regarding JSON messages

Mickbaer, 2017-02-14:
Dear Theo,
yes, there is at the FHEM Forum at this link:
https://forum.fhem.de/index.php?topic=66761.0
an discussion, where the JSON message has been splitted of to single Items using the J2R.pm.
I hope this is usefull for others.

closed 2017-03-08 by arendst