arendst/Tasmota #29: Request: Using Override.h file

tam481, 2017-02-05:
Hi Theo,
Please enable the ability to preconfigure devices using the override file as before. I don't seem to be able to configure the DHT sensor using the override file anymore.

arendst, 2017-02-06:
``user_config.h`` and ``user_config_override.h`` are soooo 2016....

They both still exist for the remaining global parameters but configuring sensors is done online by either the webpages or MQTT commands.

MQTT sensor configuration command sequence would be:
- ``modules`` - show supported modules
- ``module 4`` - select desired module functionality for a Sonoff TH (Wait for the restart)
- ``gpios`` - show supported sensor types. (DHT21 = AM2301, AM2302 = AM2321 = DHT22)
- ``gpio`` - show current defined sensors on supported GPIO pins
- ``gpio14 2`` - select sensor AM2301 (Wait for the restart)

Now if teleperiod is set to 300 every 5 minutes you should see the sensor telemetry data.

tam481, 2017-02-08:
thanks Theo. Is it expensive from a memory perspective to have these still configurable via the config override file? My two Sonoffs have set roles and I would like to be able to update them and keep the settings which I know they do anyway but I tend to do a "reset 2" after upgrading to ensure that no anomalies occur. I don't see why these options can't be hard-coded for those like myself. Also, it would allow the reduction of the attack surface should the network be compromised in anyway. Just a thought.

arendst, 2017-02-08:
Reset  2 does not clear the settigns flash area. It clears program flash only. it just defaults the settings.

Stop doing it this way with this code version or keep using the previous version.

The use of user_config.h with this version breaks the on the fly features introduced with this version.

closed 2017-02-13 by arendst