arendst/Tasmota #31: HC-SR501 PIR motion detector as switch not working as expected

mihalski, 2017-02-06:
Is POWER = OFF the intended result when motion is detected? Or is there some further configuration to the type of switch required to make this work more reliably.

And on another identical device with identical configuration and hardware I am getting mostly "correct" behaviour except that sometimes when it is triggered it wobbles from On to OFF to ON a few times changing every 4/5 seconds.

Perhaps there needs to be 30 seconds or so of "calibration" before the firmware takes any readings from a PIR sensor? That was my impression though I don't know if that just means unpredictable behaviour for the first 30 seconds or whether it sets the tone for future readings.

PS. I think it may have stabilised now.. I Manually toggled the switch to OFF over the web interface while no motion was being detected and so far so good.. Going to continue testing to see if the behaviour remains consistent.

Hogesyx, 2017-02-06:
Are you powering it with 5V or 3.3V with bypass modification?

mihalski, 2017-02-06:
I've tried both. No difference. They were mostly stable overnight, but one of them had a tiny bit of jitter.. Thinking to maybe set up some sort of camera triggering just to see if there is actual motion that I'm not aware of.

Mostly satisfied.. Though I wish there was some sort of software smoothing so that I could set the duration that the sensor reports as active and the delay between retriggering. The resistor based hardware controls just aren't good enough. I'd like to say if triggered show active for 1 minute and then don't retrigger  for another 30 seconds (or something like that).

Nayar, 2017-02-11:
I use my sensor only to detect motion then send an MQTT message. My homeassistant server then decides whether to turn on the lights or not. I get much more control

Reinhartd, 2017-02-18:
I have the same problem, every few seconds is switched from on to off. I have changed the hc-sr501 and now everything is ok. PS. My Sensor is plugged on 5V.

closed 2017-03-08 by arendst