arendst/Tasmota #32: why I fail to verify / compile?

hamwong, 2017-02-06:
I tried to move on with next gem firmware for sonoff, and I got difficult with compile,
I can't compile with modified user_config.h than try with all default and still fail
Log shows it fails at the final when linking everything together, why was that?
I got no issue to compile with Sonoff-MQTT-OTA-Arduino 3.2.13 and I just did it once again with success after fail with Tasmota, Did I miss something or did anything wrong?

Linking everything together...
"C:\Temp\arduino-1.8.0\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g  -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lib" "-LC:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/ld" "-Teagle.flash.1m64.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370/sonoff.ino.elf" -Wl,--start-group "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\sketch\sonoff.ino.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\Ticker\Ticker.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFi.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFiAP.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFiMulti.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFiSTA.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\ESP8266WiFiScan.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\WiFiClient.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\WiFiClientSecure.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\WiFiServer.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WiFi\WiFiUdp.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266HTTPClient\ESP8266HTTPClient.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266httpUpdate\ESP8266httpUpdate.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\pubsubclient\PubSubClient.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WebServer\ESP8266WebServer.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266WebServer\Parsing.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\DNSServer\DNSServer.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\ESP8266mDNS\ESP8266mDNS.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\Wire\Wire.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\NeoPixelAvr.c.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\NeoPixelEsp8266.c.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HsbColor.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HslColor.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HtmlColor.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HtmlColorNameStrings.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HtmlColorNames.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\HtmlColorShortNames.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\NeoEsp8266UartMethod.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\NeoGamma.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\NeoPixelAnimator.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\RgbColor.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\NeoPixelBus-master\internal\RgbwColor.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370\libraries\SPI\SPI.cpp.o" "C:\Users\Ham\AppData\Local\Temp\arduino_build_520370/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-LC:\Users\Ham\AppData\Local\Temp\arduino_build_520370"
c:/temp/arduino-1.8.0/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Ham\AppData\Local\Temp\arduino_build_520370/sonoff.ino.elf section `.text' will not fit in region `iram1_0_seg'

collect2.exe: error: ld returned 1 exit status

Using library Ticker at version 1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker 
Using library ESP8266WiFi at version 1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.1 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate 
Using library pubsubclient at version 2.6 in folder: C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\pubsubclient 
Using library ESP8266WebServer at version 1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
Using library ESP8266mDNS in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)
Using library Wire at version 1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire 
Using library NeoPixelBus-master at version 2.2.6 in folder: C:\Temp\arduino-1.8.0\portable\sketchbook\libraries\NeoPixelBus-master 
Using library SPI at version 1.0 in folder: C:\Temp\arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.

arendst, 2017-02-06:
Set Debug Level to None as there is to many txt in the source.

hamwong, 2017-02-06:
Do you mean Tools > Debug Level? it is none as default.

arendst, 2017-02-06:
Yes and set Tools > Debug Port to disabled too

hamwong, 2017-02-06:
I am sorry I really not understand, beside Tools > Debug port, I don't found anywhere else have debug option, if you mean the log I show above have so many line it's "Show verbose output druing: compilation"
I active it as I fail at the starting, so if I set it off, error as below

Arduino: 1.8.0 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Serial, None"

c:/temp/arduino-1.8.0/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Ham\AppData\Local\Temp\arduino_build_418820/sonoff.ino.elf section `.text' will not fit in region `iram1_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

hamwong, 2017-02-06:
Can I just compile your firmware which unzip  without modify?

arendst, 2017-02-06:
About options:
- option Tools > Debug Port  = disabled
- option Tools > Debug Level = none

Yes you can compile without modify.

hamwong, 2017-02-06:
Thanks, it works now, I didn't aware leave the port as serial will fail to compile.
Thanks for your help. may I buy you a coffee?

arendst, 2017-02-06:
Yep, send it to the Netherlands but keep it hot!

closed 2017-02-06 by arendst

hamwong, 2017-02-06:
XD how about give me your paypal account and you may buy a it nearby

RappC, 2019-06-03:
> About options:
> 
> * option Tools > Debug Port  = disabled
> * option Tools > Debug Level = none
> 
> Yes you can compile without modify.

I have the same error, but I'm not able to find the place where to adjust the configuration as per your suggestion. Can you point me into the correct direction? 
I'm using Atom

edelstahlratte, 2019-06-05:
Mhh, same here since 6.5.0 master under Atom/PlatformIO... Who else is experiencing this and what is the solution?

Thank you,

Nico.

mfw2010, 2019-06-09:
I've got the same issue on Atom/PlatformIO - can't find a fix yet with Debug options? Any ideas?

ascillato, 2019-06-09:
You can ask for help on the Tasmota support chat.

TheSDTM, 2019-06-23:
The same problem on VSCode PlatformIO and MacOS. Tasmota version 6.5.0 (same issue with 6.4.1 and 6.3.0). Compilation log:
```
Processing sonoff (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), tool-esptool 1.409.0 (4.9), framework-arduinoespressif8266 1.20300.1 (2.3.0)
Converting sonoff.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Mutichannel_Gas_Sensor> 0.0.1
|   |-- <Wire> 1.0
|-- <ESP8266WiFi> 1.0
|-- <rc-switch> 2.6.2
|-- <Adafruit GFX Library> 1.2.9
|   |-- <SPI> 1.0
|-- <ESP8266httpUpdate> 1.1
|   |-- <ESP8266HTTPClient> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <TasmotaSerial> 2.3.0
|-- <SPI> 1.0
|-- <IRremoteESP8266> 2.5.2
|-- <Ticker> 1.0
|-- <I2Cdevlib-ADS1115>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> 1.0
|-- <NewPing> 1.9.1
|-- <Adafruit SGP30 Sensor> 1.0.2
|   |-- <Wire> 1.0
|-- <Adafruit CCS811 Library> 1.0.0
|   |-- <Wire> 1.0
|-- <ArduinoJson> 5.13.4
|-- <Adafruit SSD1306> 1.1.2
|   |-- <Adafruit GFX Library> 1.2.9
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <ESP8266HTTPClient> 1.1
|   |-- <ESP8266WiFi> 1.0
|-- <TasmotaModbus> 1.1.0
|   |-- <TasmotaSerial> 2.3.0
|-- <BME680_driver-bme680_v3.5.9>
|-- <Wire> 1.0
|-- <FrogmoreScd30>
|   |-- <Wire> 1.0
|-- <Adafruit LED Backpack Library> 1.1.6
|   |-- <Wire> 1.0
|   |-- <Adafruit GFX Library> 1.2.9
|   |   |-- <SPI> 1.0
|-- <LinkedList>
|-- <Waveshare esp 2.9 inch e-paper display driver> 1.0
|   |-- <SPI> 1.0
|-- <NeoPixelBus> 2.2.9
|   |-- <SPI> 1.0
|-- <PubSubClient> 2.6
|-- <I2Cdevlib-MPU6050>
[... 293 lines cut ...]
Environment sonoff-sensors      [IGNORED]
Environment sonoff-display      [IGNORED]
Environment sonoff-BG           [IGNORED]
Environment sonoff-BR           [IGNORED]
Environment sonoff-CN           [IGNORED]
Environment sonoff-CZ           [IGNORED]
Environment sonoff-DE           [IGNORED]
Environment sonoff-ES           [IGNORED]
Environment sonoff-FR           [IGNORED]
Environment sonoff-GR           [IGNORED]
Environment sonoff-HE           [IGNORED]
Environment sonoff-HU           [IGNORED]
Environment sonoff-IT           [IGNORED]
Environment sonoff-KO           [IGNORED]
Environment sonoff-NL           [IGNORED]
Environment sonoff-PL           [IGNORED]
Environment sonoff-PT           [IGNORED]
Environment sonoff-RU           [IGNORED]
Environment sonoff-SE           [IGNORED]
Environment sonoff-SK           [IGNORED]
Environment sonoff-TR           [IGNORED]
Environment sonoff-TW           [IGNORED]
Environment sonoff-UK           [IGNORED]
============================================== 1 failed, 0 succeeded in 14.56 seconds ==============================================
```

ascillato, 2019-06-23:
Hi, please, address this to the Tasmota support chat. Thanks

dedors, 2019-06-29:
When you get lucky to meet someone that can help there...
Well I didn't and wasted 3 more hours.

In the end, I switched from 2.3 to 2.4 in plattform.ini:

platform                  = ${core_2_4_2.platform}
build_flags               = ${core_2_4_2.build_flags}

then clean first, then build.

snakuzzo, 2019-07-01:
Same problem here, with 2.3.0 core I can't build the firmware, with 2.5.2 it works...

```
Linking .pioenvs/sonoff-IT/firmware.elf
/Users/snake/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/
../../../../xtensa-lx106-elf/lib/libm.a(lib_a-ef_fmod.o):(.literal+0x0): undefined refere
nce to `__ieee754_remainderf'
/Users/snake/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/
../../../../xtensa-lx106-elf/lib/libm.a(lib_a-ef_fmod.o): In function `__ieee754_fmodf':
ef_fmod.c:(.text+0x36): undefined reference to `__ieee754_remainderf'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff-IT/firmware.elf] Error 1
========================== [ERROR] Took 48.68 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment sonoff              [SKIP]
Environment sonoff-minimal      [SKIP]
Environment sonoff-basic        [SKIP]
Environment sonoff-classic      [SKIP]
Environment sonoff-knx          [SKIP]
Environment sonoff-sensors      [SKIP]
Environment sonoff-display      [SKIP]
Environment sonoff-BG           [SKIP]
Environment sonoff-BR           [SKIP]
Environment sonoff-CN           [SKIP]
Environment sonoff-CZ           [SKIP]
Environment sonoff-DE           [SKIP]
Environment sonoff-ES           [SKIP]
Environment sonoff-FR           [SKIP]
Environment sonoff-GR           [SKIP]
Environment sonoff-HE           [SKIP]
Environment sonoff-HU           [SKIP]
Environment sonoff-KO           [SKIP]
Environment sonoff-NL           [SKIP]
Environment sonoff-PL           [SKIP]
Environment sonoff-PT           [SKIP]
Environment sonoff-RU           [SKIP]
Environment sonoff-SE           [SKIP]
Environment sonoff-SK           [SKIP]
Environment sonoff-TR           [SKIP]
Environment sonoff-TW           [SKIP]
Environment sonoff-UK           [SKIP]
Environment sonoff-IT           [ERROR]
========================== [ERROR] Took 48.69 seconds ==========================
```

ascillato, 2019-07-01:
Hi

Sorry, I can't reproduce your issue. 
Please, address this to the Tasmota support chat. Thanks

ascillato, 2019-07-12:
Anyone having an issue compiling with Platformio 4 and core 2.3.0, please read issue https://github.com/arendst/Sonoff-Tasmota/issues/6073

nightcat91, 2019-07-12:
I had the same error while trying to build using Atom and PlatformIO (pio).
Using the zip file from v. 6.6.0 

I fixed it with the help from @Jason2866 on discord.
Its a known bug which can also be found here:
https://github.com/arendst/Sonoff-Tasmota/issues/6073
It just doesnt work with the new version 4.0 of pio

A temporary fix for now is to downgrade the pio version:
Open a terminal in Atom (Alt+Shift+T) and type:
pip uninstall platformio
pip install platformio==3.6.7

Now it should build without problems!

ivankravets, 2019-07-13:
This is not a bug. See my comment here https://github.com/arendst/Sonoff-Tasmota/issues/6073#issuecomment-511109803

Jason2866, 2019-07-13:
Platformio.ini is updated. Tasmota compiles fine now with core 2.3.0 with V.4.0 of Platformio.
No special version of Platformio needed anymore