arendst/Tasmota #33: Upload Error

rocklobster42195, 2017-02-06:
Hi.
When I tried to upload the firmware, I got this error:

```
[02/06/17 19:07:46] Processing sonoff (platform: espressif8266, lib_deps: PubSubClient, NeoPixelBus, build_flags: -Wl,-Tesp8266.flash.1m64.ld -DMQTT_MAX_PACKET_SIZE=400, board: esp01_
1m, framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 25 compatible libraries
Looking for dependencies...
Library Dependency Graph

|-- <PubSubClient> v2.6
|-- <NeoPixelBus> v2.2.6
Linking .pioenvs\sonoff\firmware.elf
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff\firmware.elf] Error 1
 [ERROR] Took 3.31 seconds
```

Can anybody help me please?
Thank you,
Boris.

Mickbaer, 2017-02-07:
Hello,
I have downloaded the Tasmato V3.9.11 and compiled it without chnages on my Windows7 Platformio.
successfully. And it works great.

IDE 1.7.2 | Core 3.2.1

[02/07/17 01:42:27] Processing sonoff (platform: espressif8266, lib_deps: PubSubClient, NeoPixelBus, build_flags: -Wl,-Tesp8266.flash.1m64
.ld -DMQTT_MAX_PACKET_SIZE=400, board: esp01_1m, framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Converting sonoff.ino
Collected 25 compatible libraries
Looking for dependencies...

Library Dependency Graph
|-- PubSubClient v2.6
|-- NeoPixelBus v2.2.6
|   |-- SPI v1.0
|-- ESP8266WiFi v1.0
|-- DNSServer v1.1.0
|   |-- ESP8266WiFi v1.0
|-- ESP8266httpUpdate v1.1
|   |-- ESP8266HTTPClient v1.1
|   |   |-- ESP8266WiFi v1.0
|   |-- ESP8266WiFi v1.0
|-- ESP8266WebServer v1.0
|   |-- ESP8266WiFi v1.0
|-- Wire v1.0
|-- ESP8266HTTPClient v1.1
|   |-- ESP8266WiFi v1.0
|-- ESP8266mDNS
|   |-- ESP8266WiFi v1.0
|-- Ticker v1.0
Compiling .pioenvs\sonoff\src\sonoff.ino.o
Archiving .pioenvs\sonoff\libFrameworkArduinoVariant.a
Compiling .pioenvs\sonoff\FrameworkArduino\Esp.o
Compiling .pioenvs\sonoff\FrameworkArduino\FS.o
Compiling .pioenvs\sonoff\FrameworkArduino\HardwareSerial.o
Compiling .pioenvs\sonoff\FrameworkArduino\IPAddress.o
Compiling .pioenvs\sonoff\FrameworkArduino\MD5Builder.o
Compiling .pioenvs\sonoff\FrameworkArduino\Print.o
Compiling .pioenvs\sonoff\FrameworkArduino\Schedule.o
Compiling .pioenvs\sonoff\FrameworkArduino\Stream.o
Compiling .pioenvs\sonoff\FrameworkArduino\StreamString.o
Compiling .pioenvs\sonoff\FrameworkArduino\Tone.o
Compiling .pioenvs\sonoff\FrameworkArduino\Updater.o
Compiling .pioenvs\sonoff\FrameworkArduino\WMath.o
Compiling .pioenvs\sonoff\FrameworkArduino\WString.o
Compiling .pioenvs\sonoff\FrameworkArduino\abi.o
Compiling .pioenvs\sonoff\FrameworkArduino\base64.o
Compiling .pioenvs\sonoff\FrameworkArduino\cbuf.o
Compiling .pioenvs\sonoff\FrameworkArduino\cont.o
Compiling .pioenvs\sonoff\FrameworkArduino\cont_util.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_eboot_command.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_flash_utils.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_i2s.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_main.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_noniso.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_phy.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_postmortem.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_si2c.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_timer.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring_analog.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring_digital.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring_pulse.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring_pwm.o
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_wiring_shift.o
Compiling .pioenvs\sonoff\FrameworkArduino\debug.o
Compiling .pioenvs\sonoff\FrameworkArduino\heap.o
[... 43 lines cut ...]
Compiling .pioenvs\sonoff\lib\ESP8266WiFi\WiFiUdp.o
Compiling .pioenvs\sonoff\lib\DNSServer\DNSServer.o
Compiling .pioenvs\sonoff\lib\ESP8266HTTPClient\ESP8266HTTPClient.o
Compiling .pioenvs\sonoff\lib\ESP8266httpUpdate\ESP8266httpUpdate.o
Archiving .pioenvs\sonoff\lib\libDNSServer.a
Archiving .pioenvs\sonoff\lib\libESP8266WiFi.a
Compiling .pioenvs\sonoff\lib\ESP8266WebServer\ESP8266WebServer.o
Compiling .pioenvs\sonoff\lib\ESP8266WebServer\Parsing.o
Archiving .pioenvs\sonoff\lib\libESP8266httpUpdate.a
Compiling .pioenvs\sonoff\lib\Wire\Wire.o
Archiving .pioenvs\sonoff\lib\libWire.a
Archiving .pioenvs\sonoff\lib\libESP8266HTTPClient.a
Compiling .pioenvs\sonoff\lib\ESP8266mDNS\ESP8266mDNS.o
Compiling .pioenvs\sonoff\lib\Ticker\Ticker.o
Archiving .pioenvs\sonoff\lib\libESP8266WebServer.a
Archiving .pioenvs\sonoff\lib\libTicker.a
Archiving .pioenvs\sonoff\lib\libESP8266mDNS.a
Linking .pioenvs\sonoff\firmware.elf
Calculating size .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
text       data     bss     dec     hex filename
445315    10684   35136  491135   77e7f .pioenvs\sonoff\firmware.elf
 [SUCCESS] Took 48.03 seconds

I hope this will help you.

rocklobster42195, 2017-02-07:
How embarrassing! I put the `platformio.ini` in the wrong folder. It´s working now.
Thank you,
Boris

closed 2017-02-07 by arendst

petonic, 2018-02-26:
@rocklobster42195 -- what directory does it go in and which directory did you have it in?  I'm running into the same problem. Thanks!