arendst/Tasmota #34: Not getting any Temp data from Tasmota

citrusfizz, 2017-02-06:
Updated to Tasmota (first version and now todays version)
and i can't get any data for my Temp/Humidity sensor. 
I turned on 4 debug logging and i don't see anything that says its even trying to get the data. 
it is a dht22 sensor
i have configured gpio 14 for dht22  in the web interface and tried to set the other two GPIO options the same.  tried am3201 as the device.   
tried setting the module as a Sonoff TH instead of basic. 
nothing give me any temp data. 

not sure what else to do to get data like i did in the previous git hub project. 
here is my user_config.h   minus the irrelevant parts. 
http://pastebin.com/xfdVWN29

thanks

arendst, 2017-02-06:
Only one gpio may be configured for DHT/AMxxx so that should be gpio14. Set the others to none.

See wiki > first steps regarding sensor config.

A power cycle after configuration might help.

citrusfizz, 2017-02-07:
I guess i wasn't to clear,  i know that gpio14 is the one that needs to be set,  i just set the others in hopes something might trigger it to work.   i tried every combination with no luck.  I read the wiki however there is nothing in the hardware modification. Power cycled many times, each while making a change,  unlike the older firmware,  this one does not even output that its trying to get data.  the older firmware would say something like CRC error, or No data available.   this one doesn't appear to have anything.   thats why i posted my config.h and the console output.

arendst, 2017-02-08:
OK let's start off from the beginning.

Without any sensor connected but with weblog set to 3 you should see at least the following lines after a restart:
```
00:00:00 Config: Loaded configuration from flash at F8 and count 117
00:00:00 APP: Bootcount 53
00:00:00 DHT: Max clock cycles 80000
00:00:00 APP: Project sonoff th1 (Topic th1, Fallback DVES_07C8FF, GroupTopic sonoffs) Version 3.9.12
```
The third line is from the DHT code calibrating it's timer. I do not see it in your log so apparently you did non configure any GPIO for a DHT sensor.

Look in the wiki > Sensor Configuration and retry.
[image]

citrusfizz, 2017-02-10:
@arendst I flashed the latest 3.9.13  and sensor is working fine again.  No other changes were made. 
Greatful for everything you have done with this project.   Cheers.

closed 2017-02-13 by arendst