arendst/Tasmota #35: espcomm_open failed; espcomm_upload_mem failed

binderth, 2017-02-07:
I'm trying to flash a Sonoff POW with 3.9.11a, doing these steps:

1. preparing the sonoff (connecting 3.3V, GND, TX-RX, RX-TX)
2. not connecting 230V! ;)
3. connecting the "Yongse – FT232RL FTDI USB zu TTL Serial Adapter Modul" with my Win10-USB
4. creating a directory and "`git clone https://github.com/arendst/Sonoff-Tasmota`"
5. "open Folder" in PlatformIO (running on python 2.7)
6. PlatformIO: Build (works without a problem)
7. PlatformIO: Upload

now I get this errors in my log:
```
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0

Compiling .pioenvs\sonoff\src\sonoff.ino.o
Linking .pioenvs\sonoff\firmware.elf
Checking program size
text       data     bss     dec     hex filename
445299    10668   35136  491103   77e5f .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Looking for upload port...
Auto-detected: COM5
Uploading .pioenvs\sonoff\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 17.86 seconds
 ```

What am I doing wrong?

binderth, 2017-02-07:
Oh never mind! Problem lay in the "flash mode" - step 7 in my list.
It isn't going to work that way. After re-reading the HowTo, you must Change step 7 to:

7.1 disconnecting the USB and pressing the button on the Sonoff POW
7.2 connecting the USB while pressing the button => LED lights (without blinking)

then, the Upload works just fine.

closed 2017-02-07 by binderth

jaysuk, 2017-02-07:
>    Sonoff-Tasmota"

Did you hold down the button on the sonoff before plugging in the FTDI (or
connecting an external power supply) to put the sonoff in programming mode?

binderth, 2017-02-07:
yes, that was it - found it out same time! ;)

andreasloe, 2017-08-12:
I hold down the button BEFORE plugging in the FTDI. Then I connected the external power (read USB), still holding the button down. No LED went on. Then I started "uploading" from arduino, everything was built but I still got:

Arduino: 1.8.1 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), ck, Disabled, None"

Arduino: 1.8.1 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), ck, Disabled, None"

Der Sketch verwendet 483396 Bytes (47%) des Programmspeicherplatzes. Das Maximum sind 1023984 Bytes.
Globale Variablen verwenden 43196 Bytes (52%) des dynamischen Speichers, 38724 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
/Users/xxxxxxxxx/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd ck -cb 115200 -cp /dev/cu.usbserial-A9RCE5PE -ca 0x00000 -cf /var/folders/l6/myrckv8d12z9_06fl49fm36h0000gn/T/arduino_build_638433/sonoff.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A9RCE5PE
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A9RCE5PE at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
Beim Hochladen des Sketches ist ein Fehler aufgetreten
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

davidelang, 2017-08-13:
check that you don't have the tx/rx lines reversed, and that you are using the 
correct com port. this look like you are never 
connecting to the device.
k

andreasloe, 2017-08-14:
Yes, I made the terrible mistake of connecting TX-TX and RX-RX. Sorry for that. I just now read how UART works.