arendst/Tasmota #36: Crash by click on "Configure Wifi"

jdpiguet, 2017-02-07:
Hi Theo,

I got a crash in a Sonoff Basic.
The Wifi definition in the user_config.h:
```
// -- Wifi ----------------------------------------
#define STA_SSID1              "XiaoMI"          // [Ssid1] Wifi SSID
#define STA_PASS1              "Xiaomi110"       // [Password1] Wifi password
#define STA_SSID2              "sonoff"          // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "YjvUodcPU0NHu2"  // [Password2] Optional alternate AP Wifi password
#define WIFI_HOSTNAME          "%s%04d"          // [Hostname] Expands to <MQTT_TOPIC>-<last 4 decimal chars of MAC address>
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY)
```

If you need the Wifi password, I can send it in PM.

Thanks for your work, you owe more than one beer!
Regards, Jacques-D.

```
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
▒
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_95FD98, GroupTopic sonoffs) Version 3.9.11
00:00:00 Wifi: Connecting to AP1 XiaoMI in mode 11N as sonoff-7576...
00:00:09 Wifi: Connected
00:00:09 HTTP: Webserver active on sonoff-7576 with IP address 10.211.122.191
11:16:54 MQTT: Attempting connection...
11:16:54 MQTT: Connected
11:16:54 MQTT: tele/sonoff/LWT = Online (retained)
11:16:54 MQTT: cmnd/sonoff/POWER =
11:16:54 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"3.9.11", "FallbackTopic":"DVES_95FD98", "GroupTopic":"sonoffs"}
11:16:54 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-7576", "IPaddress":"10.211.122.191"}
11:16:55 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
11:16:55 MQTT: stat/sonoff/POWER = OFF
11:17:02 MQTT: tele/sonoff/STATE = {"Time":"2017-02-07T11:17:02", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"XiaoMI", "RSSI":76}}

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000002e8 depc=0x00000000

ctx: cont
sp: 3fff1fe0 end: 3fff2490 offset: 01a0

>>>stack>>>
3fff2180:  00000003 06a14ca5 00000000 4000050c
3fff2190:  402324de 4010346c 3fffc200 00000022
3fff21a0:  00000010 000002e8 00000000 3fff2230
3fff21b0:  3fff2240 40107b8c 3fff50dc 3fff22e0
3fff21c0:  4025b6a5 00000021 3fff22e0 401004d8
3fff21d0:  3fff137c 000004bd 000004bd 4021b968
3fff21e0:  00000a00 3fff5e7c 3fff540c 4021bcd5
3fff21f0:  3fff2240 3fff2230 0000000c 7fffffff
3fff2200:  000006e0 3fff2328 3fff2260 3fff2260
3fff2210:  00000002 00000001 00000000 4021bd0c
3fff2220:  3fff2240 3fff2230 0000000c 4021af0f
3fff2230:  00000002 00000001 3fff2328 000002e8
3fff2240:  00000000 4025ee6e 00000000 3fff2260
3fff2250:  00000002 00000001 00000000 402102b8
3fff2260:  00000000 00000000 00000000 401023f2
3fff2270:  3ffea745 401056cf 3ffee1d0 401023f2
3fff2280:  40103105 3ffee1d0 00000002 401023f2
3fff2290:  00007fff 06d80d23 3ffeebbc 40103294
3fff22a0:  3fff137c 000001ea 000001ea 3fff137c
3fff22b0:  00000010 3fff2380 00000000 4010053d
3fff22c0:  00000010 3fff2380 00000000 4010053d
3fff22d0:  3fff137c 00000203 00000203 4010020c
3fff22e0:  00000000 00000000 00000000 3fff137c
3fff22f0:  00000010 3fff2380 00000000 4010053d
3fff2300:  40248316 00000000 3fff2380 4024832d
3fff2310:  3fff3e3c 3fff2380 3fff2380 3ffe93f2
3fff2320:  00000010 3fff2380 3fff5e7c 000009ff
3fff2330:  000009f5 3fff23c0 00000000 4010053d
3fff2340:  3fff137c 000001d6 000001d6 3fff137c
3fff2350:  00000010 3fff23e0 00000000 40217ec8
3fff2360:  00000010 3fff23c0 3fff23c0 4021aec0
3fff2370:  3fff3544 00000003 3fff23c0 40217ec8
3fff2380:  00000002 00000001 3fff31dc 40210478
3fff2390:  00000002 00000001 3fff23c0 4021bdda
3fff23a0:  00000000 00000000 3fff31dc 40217ebe
3fff23b0:  3fff31dc 3fff42e4 3fff31dc 40217efa
3fff23c0:  00000000 00000000 00000000 4021b070
3fff23d0:  3fff31dc 3fff42e4 3fff42a4 40217f8d
3fff23e0:  3fff3544 0000000f 00000003 40214b8c
3fff23f0:  00000000 00000000 00000000 00000001
3fff2400:  00000002 4021742c 0000000e 3fff1470
3fff2410:  00000000 00000000 3fff0350 3fff1460
3fff2420:  00000001 3fff42c8 3fff42a4 40218117
3fff2430:  3ffe9508 00000000 3fff1470 00000000
3fff2440:  3fffdad0 3fff1460 4021bb0c 3fff1470
3fff2450:  3fffdad0 00000000 3fff1458 40203e96
3fff2460:  00000000 00000000 00000001 40213a64
3fff2470:  3fffdad0 00000000 3fff1458 4021bb58
3fff2480:  feefeffe feefeffe 3fff1470 40100718
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d

```

hoekbrwr, 2017-02-07:
I just took out the last 4 digit from the hostname in the config.h and with the same firmware I cannot change the Wifi config anymore! I could not configure through the webserver too and then I thought I change that in the user_confi.h and recompile and upgrade. The upgrade went well, but now I am unable to  go into the webserver's config Wifi anymore! I did **not** change to the latest version on Github in the mean time!

arendst, 2017-02-07:
I can reproduce exception (28) when opeing the Wifi Config page. Will investigate...

arendst, 2017-02-07:
I'll remove the WIFI_HOSTNAME setting from user_config.h as the goal of this version is to make changes online if possible anyway. The user can still change the hostname using the Wifi Config page and/or the ``hostname`` command.

Background: I need to write a lot of code to handle (bad) user input regarding % signs in names. As you noticed removing the dash is disastrous as the standard procedure for replacing the %s%04d by the Topic name filled with four characters from the unique MAC address fails.

hoekbrwr, 2017-02-08:
I could not get into the wifi configuration menu on the webserver(used the hostname cmnd on all my sonoff units without special characters). This was solved by doing a firmware upgrade with just the same firmware as was already loaded. Something that might help finding out when investigating the cause of this problem.

arendst, 2017-02-08:
By removing WIFI_HOSTNAME from the user_config.h and not allowing any string formatting by users the problem should have been solved.

You're still able to configure your hostname by command and webpage.

starfish1107, 2017-02-08:
I am customizing my config using user_config_override.h - I think this is the idea behind isn´t it?  will you remove hostname config possitility from this file too?

arendst, 2017-02-08:
Yes. The ability to configure hostname via user_config or user_config_override is removed.

The default hostname will be unique and derived from MQTT topic and the last four decimal digits of the MAC address like ``sonoff-1234``.

The user can then change this hostname via the hostname command or the webpage.

hoekbrwr, 2017-02-11:
There is still a #define WIFI_HOSTNAME          "%s-%04d" in sonoff.ino on line 61. This still gives the opportunity to misconfigure the name! Shouldn"t you hardcode this as a default?
I found this out when I did a try to set all values to user_config.h and got the 4 decimals again. Though that is fully OK for a clean start, the ability to tinker with it you said you did not like.

arendst, 2017-02-11:
I know. But if someone wants to change it outside user_config.h I won't fix any problem that comes out of it.

closed 2017-02-13 by arendst