arendst/Tasmota #37: Two Sonoff Pows playing Ping-Pong

binderth, 2017-02-07:
I flashed two Sonoff Pows with the Tasmota Firmware. If both are online, they seem to Play Ping-Pong with each other! ;)

`mosquitto_sub  -v -t 'tele/#' -t 'stat/#'` gives me this:
```
tele/sonoff_DiWa/STATE {"Time":"2017-02-07T20:13:42", "Uptime":1, "POWER":"ON", "Wifi":{"AP":1, "SSID":"xxx", "RSSI":56}}
tele/sonoff_DiWa/ENERGY {"Time":"2017-02-07T20:13:42", "Yesterday":"0.000", "Today":"0.002", "Period":0, "Power":0, "Factor":"0.00", "Voltage":247, "Current":"0.000"}
tele/sonoff_WaMa/LWT Online
tele/sonoff_DiWa/LWT Online
tele/sonoff_WaMa/LWT Online
tele/sonoff_DiWa/LWT Online
tele/sonoff_WaMa/LWT Online
tele/sonoff_DiWa/LWT Online
tele/sonoff_WaMa/LWT Online
tele/sonoff_WaMa/STATE {"Time":"2017-02-07T20:13:52", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"xxx", "RSSI":68}}
tele/sonoff_WaMa/ENERGY {"Time":"2017-02-07T20:13:52", "Yesterday":"0.000", "Today":"0.000", "Period":0, "Power":0, "Factor":"0.00", "Voltage":0, "Current":"0.000"}
tele/sonoff_DiWa/LWT Online
tele/sonoff_WaMa/LWT Online
tele/sonoff_DiWa/LWT Online
...
```
I set `TelePeriod 20`, which is working fine on each device, but the "LWT Online"-messages are disturbing me. If one of the Sonoff Pows is offline, there's no "LWT Online" message at all.

is there a way to tell them to stop this? ;)

arendst, 2017-02-07:
And they both have different mqtt client ids?

binderth, 2017-02-08:
Thanks! That was my problem. I did use the same user_config.h and forgot to change the ID.
Thanks!

closed 2017-02-08 by binderth