arendst/Tasmota #39: Feature request: NeoPixel Fire Effect

rocklobster42195, 2017-02-08:
How about implementing a NeoPixel Fire Effect, e.g. [http://www.walltech.cc/neopixel-fire-effect/](http://www.walltech.cc/neopixel-fire-effect/)?

arendst, 2017-02-08:
How about ``scheme 9`` ?

hoekbrwr, 2017-02-08:
Why not replace it. It is an astonishing good simulation of fire, scheme 9 is very dull!

arendst, 2017-02-08:
As soon as I see a delay in a piece of code I get weary...

hoekbrwr, 2017-02-08:
So use setTimeout and put the for loop in a at random time(50-150ms) called function? That will work?

davidelang, 2017-03-27:
@hoekbrwr it can't be anything that requires a loop for just this functionality, it needs to be something that can run in the main loop along with everything else that's happening

don-willingham, 2017-07-30:
Yes. When the software is delaying, it's not doing anything else. A module could be added to the main loop, and set a time when it will run time. If on it's next iteration not enough time has elapsed, it will simply return.  I just started digging into the sensor code, as I submitted [PR658](https://github.com/arendst/Sonoff-Tasmota/pull/658), the library I pulled from has a delay(500), ugly!

ascillato2, 2018-07-10:
Duplicated issue. Closing. Leaving #1952 open.

closed 2018-07-10 by ascillato2