arendst/Tasmota #40: Mismatch of GPIO3/4 with Sonoff SV calling it GPIO4/5

hoekbrwr, 2017-02-08:
I am confused now. The pins GPIO4 and 5 are called GPIO3 and 4 in your template. I connected a LED WS2812 string to GPIO4 and configured that, works very nice. I noticed @mihalski with his Huzzah Feather defined the template with GPIO4 and 5 available for that module. 
On the board it is called 4 and 5!
What is going on here, I am obviously missing something?

arendst, 2017-02-08:
Yes. You're missing a flag I forgot to enter.

Quick workaround: update sonoff_template.h line 111 from
```
0, 0, 0, 0, 0, 0, 0,
```
to
```
GPIO_USER,        // GPIO05 Optional sensor
0, 0, 0, 0, 0, 0,
```

hoekbrwr, 2017-02-08:
What about line 109? GPIO3 is not there!

arendst, 2017-02-08:
GPIO03 is the RXD port and some people are using it as it is easily accessible by the programming header.

closed 2017-02-11 by arendst