arendst/Tasmota #41: Using Sonoff SV for dimming Power LEDs

hoekbrwr, 2017-02-08:
Can I use Sonoff SV to dimming a set of 12V LEDs from one of the outputs 4,5 or 14? It works on the WS2812, but that is all dimming 3 colors as I had a peek in your source code. Of course I'II have to put a LED driver in too, but that could even be integrated over the contacts of the relay and us full on(relay) and dimmed when relay off and PWM out from the "dimmer".

arendst, 2017-02-08:
It won't work as a WS2812 defined GPIO.

It might work as a Sonoff LED which uses GPIO4, 5 and 13 for GRB led drivers.

hoekbrwr, 2017-02-08:
I will have a try!

don-willingham, 2017-03-04:
I just ran across this: http://www.cnx-software.com/2017/03/03/ai-light-esp8266-based-wifi-rgb-light-bulb-supports-mqtt-via-espurna-open-source-firmware/

closed 2017-03-08 by arendst