arendst/Tasmota #43: Help needed adding MQTT_GRPTOPIC to all topics

SrKlein86, 2017-02-08:
Hello, I'd like to know if there is any way to achieve (in a relatively easy way) an extra level on my MQTT topics:
I have to deploy sensors on a 3 floor building, some of them are sonoffs, some other are not.
To make everything easier for me, I decided to make everything follow the cmnd, stat, and tele scheme, but now I realize i should have something that differenciates my sonoffs from other sensors, but still be able to tell one sonoff from the other.
What I want to achieve is a topic structure like this:
MQTT_GRPTOPIC/SUB_PREFIX/MQTT_TOPIC
MQTT_GRPTOPIC/PUB_PREFIX/MQTT_TOPIC
MQTT_GRPTOPIC/PUB_PREFIX2/MQTT_TOPIC/UPTIME, POWER/LIGHT and TIME

So I can subscribe to (for example): 3FSonoffs/tele/+/uptime (where MQTT_TOPIC is unique for each sonoff on the third floor)

I have tried "patching" this on user config, by adding concatenating like this:
#define PUB_PREFIX2            MQTT_GRPTOPIC "/tele"
I succesfully send mqtt messages with the topic structure I want, but the problem comes when I try to access the console web interface: I just get a blank console where there used to be a lot of useful info.

Can anyone help me?

SrKlein86, 2017-02-23:
Been taking a look to the code, but I can't figure out why is this happening.
I've been searching in webserver.ino and sonoff files, withouth success. The code is too complex for me, and I am unable to find anything that relates topics and console interface.

lucwuyts, 2017-02-24:
I have tried this with these defines

#define SUB_PREFIX             "test/cmnd"
#define PUB_PREFIX             "test/stat"            
#define PUB_PREFIX2            "test/tele" 

And the console shows normal messages:
15:56:05 MQTT: Connected
15:56:10 RSLT: test/tele/sonoff/LWT = Online
15:56:15 RSLT: test/cmnd/sonoff/POWER = 
15:56:27 RSLT: test/tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"3.9.20", "FallbackTopic":"DVES_82579D", "GroupTopic":"sonoffs"}
15:56:27 RSLT: test/tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-6045", "IPaddress":"10.0.250.72"}
15:56:28 MQTT: Attempting connection...
15:56:34 RSLT: test/stat/sonoff/RESULT = {"POWER":"OFF"}
15:56:34 RSLT: test/stat/sonoff/POWER = OFF

SrKlein86, 2017-02-27:
Thanks for your reply.
I've downloaded the code again in order to have it completely unmodified.
After that, I only changed network and mqtt parameters so my Sonoff could connect to my mqtt server, leaving default topics, just to test.
To my surprise, the device is already configured (the first time i flashed it, I had to choose which kind of Sonoff it was, DHT in my case, from the web interface) and still nothing shows on the console screen.
Maybe this has something to do? I thought the whole memory was wiped when flashing, and it looks like theres some remanent config that isn't.

davidelang, 2017-02-27:
when you do an over-the-air flash, it retains as much of the old config as 
possible.

If you do a flash via serial it should need to be reconfigured from scratch.

David Lang

SrKlein86, 2017-02-27:
Well.. I'm flashing it via serial, and it does look like it retains config :/

lucwuyts, 2017-02-27:
If you want to force an update to the parameters, increment  CFG_HOLDER

SrKlein86, 2017-02-27:
I've changed wifi SSID in userconfig.h just to test, and started monitoring the serial port for device info.
Just after flashing again, it just connects to the previously configured SSID instead of the new one.
@lucwuyts Thanks for the tip, will try.

altelch, 2017-02-27:
Thats why you should change CFG_HOLDER. Config changes in the code are not honoured if CFG_HOLDER doesn't change. The Sonoff then doesn't recognize the change in the defaults and still uses the saved parameters.

SrKlein86, 2017-02-27:
OK, thanks everyone so much! that was what caused the issue I was experiencing!
Just one more question before closing this thread: Is there any way to set what kind of module and sensor I'm using before flashing? i'll have to deploy a bunch of DHTs with AM2301s without web interface, and it feels tedious to configure one by one via MQTT messages.

closed 2017-03-07 by SrKlein86