arendst/Tasmota #45: Request: dynamic relays and keys

aferraz, 2017-02-08:
Hi,
 this is a great firmware, my congratulations.
I wanted to ask: why not also make the relays and keys configurable from the web page ?

Best Regards
 Alessandro

arendst, 2017-02-08:
Too much variables I have to check to keep it functional.

aferraz, 2017-02-08:
@arendst thanks you for the response, I understand your point.
Instead the possibility to read an analog sensor ?
I'm using your firmware on nodemcu and it works perfectly, it might be convenient to be able to connect other types of sensors or, for example, a tft display. 
Maybe it is already possible but I do not know how to do.

Best Regards
 Alessandro

mihalski, 2017-02-10:
A TFT display isn't a sensor.. It's an output device.

I expect that for the time being this sort of thing is probably outside the scope of this firmware. Especially with its intent to target Sonoff devices which have very limited storage and memory unlike more generic ESP8266 modules which greatly vary in those aspects.

closed 2017-03-06 by aferraz