arendst/Tasmota #46: Sonoff TH not working properly

smadds, 2017-02-09:
I have a Sonoff TH that has starting behaving strangely. 2 indicators:
1. The web server returns ERR_CONTENT_LENGTH_MISMATCH
2. Any MQTT command sent to it returns stat/......../COMMANDS1 to COMMAND4 reminders of which commands are available.

It's not very accessible, so before getting ladders to power cycle it is there anything else I can try?

arendst, 2017-02-09:
I guess this doesn't work either:
```
http://sonoff/cm?cmnd=restart%201
```
So all you can do is cycle power.

How long did it perform well before starting go haywire?

arendst, 2017-02-10:
Are you using weblog x? If so it might log too long messages depending on the text  string length which depends on the amount of sensors connected. weblogs are stored in (tight) memory.

Perhaps you can disable weblog and install a linux syslog server for remote logging using syslog x.

Your networkproblem may be related to a gateway problem on sonoff. Why, I don't know but may be memory related too. Look at the information webpage lower part about amount of free memory.

So I suggest to disable weblog for the moment and track performance.

smadds, 2017-02-10:
OK, will do.

I also think that having "/" characters in the mqtt topic is causing problems. By removing them we have restored stability to the rogue TH

arendst, 2017-02-10:
It might also be related to the sleep functionality that although it was set to 0 it still put the wifi to light sleep.

Solved in version 3.9.13 released today..

closed 2017-02-15 by arendst