arendst/Tasmota #49: Enhancement request - relay on/off to accept boolean true/false

smadds, 2017-02-10:
Could the alternative options for switching the relay(s) on and off include true/false so it will work with a Boolean input?

(I am looking to use a ramp thermostat - http://flows.nodered.org/node/node-red-contrib-ramp-thermostat and it would save having to add converters)

arendst, 2017-02-10:
You mean 0 for false and 1 for true? If so than that's already implemented. Power 0 is off, Power 1 is on, Power 2 is toggle.

smadds, 2017-02-10:
In Node Red a Boolean output is sent as string "true" or string "false", so
could those be aliases for 1 and 0?

ᐧ

arendst, 2017-02-10:
Ah, can't you put a function node behind the node putting out true/false and convert it to 1/0?

smadds, 2017-02-10:
Yes, of course. Just thought that it would increase the flexibility of the
Tasmota code and make the node-red flows easier to follow.

arendst, 2017-02-11:
Fixed in version 3.9.14 just released

smadds, 2017-02-11:
Brilliant. Thanks Theo

closed 2017-02-13 by arendst