arendst/Tasmota #50: high latency

starfish1107, 2017-02-10:
just noticed high latency time after changing to Tasmota 3.9.11. When issuing power commands it takes up to 2 seconds before relay is switching. mqtt-spy shows command transmitted in time. Sporadically it turns into loop with fast flashing green led on sonoff th10 and restarting after stopped wifi comm. Sri had no time yet for further investigations but wonder if this problem is known and/or I am misconfiguring something ... (same config as in Sonoff-MQTT-OTA-Arduino 3.2.13 btw.  where everything was OK).

arendst, 2017-02-10:
I noticed it too. I thought it was my MQTT broker implementation but as you notice it too I need to investigate

arendst, 2017-02-10:
Do you have domoticz idx configured? I notice that if I disable domoticz (by setting Idx 1 to 0) it has no latency.

starfish1107, 2017-02-10:
I disabled define USE_DOMOTICZ in user_config.h, is this OK?  - where do I find Idx 1  you mentioned ?

arendst, 2017-02-10:
It's in the Configuration > Configure Domoticz webpage or use command ``domoticzidx1 0``. But from what you're saying I guess you're not using domoricz do you?

starfish1107, 2017-02-10:
right, I am not using Domoticz - but iobroker. I´ll try disabling the way you said.

arendst, 2017-02-10:
No need to. It's probably set to zero anyway and it seems to be more caused by lack of free RAM. Version 3.9.2 is also latency free...

arendst, 2017-02-10:
It's probably related to the sleep function. You could try it yourself by performing the followings steps:
1.  ``sleep 0`` - to disable sleep mode if it was enabled
2.  change code in file support.ino line 425 to
```
//  WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
```
by adding two slashes in front to disable sleep altogether.
Recompile and give it a try.

starfish1107, 2017-02-10:
the domoticz menue does not exist here.  OK on 3.9.2   but this does not support * Advertise command topic to be used by iobroker (#299) - which was the main reason I switched to Tasmota.

starfish1107, 2017-02-10:
switched off sleepmode - now it reacts reasonably fast.  Thanks.

arendst, 2017-02-10:
Just fixed it in a nice way in version 3.9.13 just released. Thnx for your cooperation.

closed 2017-02-10 by arendst