arendst/Tasmota #51: How do I add another GPIO function

altelch, 2017-02-10:
Hi,

fumbled something during testing yesterday. Now I found that if setting GPIO14 to my newly assigned function (sonoff_template.h) the system reboots and after reboot says it switches to default configuration and reboots again which invalidates my changed GPIO_USER function. I added a new Function between GPIO_SENSOR_START and GPIO_SENSOR_END, incremented GPIO_SENSOR_END, increased the second sensors[][] index and added my function name. I can see the new function assigneable to GPIO_USER in the webfront but after assigning I get 2 reboots and after the second reboot everything is on config_default.
What would be the supposed way to add another function?

Sorry to bother,

             Heiko.

arendst, 2017-02-10:
I noticed you did add an ir remote device but the way you did it is not complete. Take a look at this:
```
#define GPIO_SENSOR_START  0
#define GPIO_NONE          0    // Not used
#define GPIO_DHT11         1    // DHT11
#define GPIO_DHT21         2    // DHT21
#define GPIO_AM2301        2    // AM2301
#define GPIO_DHT22         3    // DHT22
#define GPIO_AM2302        3    // AM2302
#define GPIO_AM2321        3    // AM2321
#define GPIO_DSB           4    // Single wire DS18B20 or DS18S20
#define GPIO_I2C_SCL       5    // I2C SCL
#define GPIO_I2C_SDA       6    // I2C SDA
#define GPIO_WS2812        7    // WS2812 Led string
#define GPIO_IRSEND        8    // Send IR codes
#define GPIO_SWT1          9    // User connected external switches
#define GPIO_SENSOR_END    10

#define GPIO_SWT2          GPIO_SENSOR_END
#define GPIO_SWT3          GPIO_SENSOR_END +1
#define GPIO_SWT4          GPIO_SENSOR_END +2
#define GPIO_KEY1          GPIO_SENSOR_END +3   // Button usually connected to GPIO0
#define GPIO_KEY2          GPIO_SENSOR_END +4
#define GPIO_KEY3          GPIO_SENSOR_END +5
#define GPIO_KEY4          GPIO_SENSOR_END +6
#define GPIO_REL1          GPIO_SENSOR_END +7   // Relays
#define GPIO_REL2          GPIO_SENSOR_END +8
#define GPIO_REL3          GPIO_SENSOR_END +9
#define GPIO_REL4          GPIO_SENSOR_END +10
#define GPIO_REL1_INV      GPIO_SENSOR_END +11
#define GPIO_REL2_INV      GPIO_SENSOR_END +12
#define GPIO_REL3_INV      GPIO_SENSOR_END +13
#define GPIO_REL4_INV      GPIO_SENSOR_END +14
#define GPIO_LED1          GPIO_SENSOR_END +15  // Leds
#define GPIO_LED2          GPIO_SENSOR_END +16
#define GPIO_LED3          GPIO_SENSOR_END +17
#define GPIO_LED4          GPIO_SENSOR_END +18
#define GPIO_LED1_INV      GPIO_SENSOR_END +19
#define GPIO_LED2_INV      GPIO_SENSOR_END +20
#define GPIO_LED3_INV      GPIO_SENSOR_END +21
#define GPIO_LED4_INV      GPIO_SENSOR_END +22
#define GPIO_PWM0          GPIO_SENSOR_END +23  // Cold
#define GPIO_PWM1          GPIO_SENSOR_END +24  // Warm
#define GPIO_PWM2          GPIO_SENSOR_END +25  // Red (swapped with Blue from original)
#define GPIO_PWM3          GPIO_SENSOR_END +26  // Green
#define GPIO_PWM4          GPIO_SENSOR_END +27  // Blue (swapped with Red from original)
#define GPIO_RXD           GPIO_SENSOR_END +28  // Serial interface
#define GPIO_TXD           GPIO_SENSOR_END +29  // Serial interface
#define GPIO_HLW_SEL       GPIO_SENSOR_END +30  // HLW8012 Sel output (Sonoff Pow)
#define GPIO_HLW_CF1       GPIO_SENSOR_END +31  // HLW8012 CF1 voltage / current (Sonoff Pow)
#define GPIO_HLW_CF        GPIO_SENSOR_END +32  // HLW8012 CF power (Sonoff Pow)
#define GPIO_USER          GPIO_SENSOR_END +33  // User configurable
#define GPIO_MAX           GPIO_SENSOR_END +34
```
That should work better.

altelch, 2017-02-10:
Ok, didn't get I have to increment the non sensor related defines, too. In the original code all values where hard coded. This version is easier to understand. Changed my code to the above version and changed
  const char sensors[GPIO_SENSOR_END][10] PROGMEM =
    { "None", "DHT11", "AM2301", "DHT22", "DS18x20", "I2C SCL", "I2C SDA", "WS2812", "IR Remote", "Switch" }; 
to reflect the above ordering. Recompiled, flashed, changed in the webfrontend and get:

HTTP: Basic Sonoff Module, GPIO3 0, GPIO4 0, GPIO14 8

after that restart wth boot_mode(3,6). During restart again Config: Use default configuration
another restart boot_mode(3,6) and the sonoff is in initial mode again.

Just pushed the current code version to make checking easier.

Thanks,

      Heiko.

arendst, 2017-02-10:
My change using GPIO_SENSOR_END doesn't work. I will investigate

arendst, 2017-02-11:
Changed in version 3.9.14 just released

altelch, 2017-02-11:
Hi,

same behaviour as before. Loggs config, restart, default config, restart.

I did:

// User selectable GPIO functionality
enum upins_t {
  GPIO_NONE,           // Not used
  GPIO_DHT11,          // DHT11
  GPIO_DHT21,          // DHT21, AM2301
  GPIO_DHT22,          // DHT22, AM2302, AM2321
  GPIO_DSB,            // Single wire DS18B20 or DS18S20
  GPIO_I2C_SCL,        // I2C SCL
  GPIO_I2C_SDA,        // I2C SDA
  GPIO_WS2812,         // WS2812 Led string
  GPIO_IRSEND,				 // IR remote
  GPIO_SWT1,           // User connected external switches
  GPIO_SENSOR_END };

// Text in webpage Module Parameters and commands GPIOS and GPIO
const char sensors[GPIO_SENSOR_END][10] PROGMEM = {
  "None",
  "DHT11",
  "AM2301",
  "DHT22",
  "DS18x20",
  "I2C SCL",
  "I2C SDA",
  "WS2812",
  "IR remote",
  "Switch" };

Greetings,

          Heiko.

altelch, 2017-02-11:
Hi, ok, your code is ok. I called the begin function of IRremoteESP8266 in GPIO-Init if the GPIO was set.
That caused the fallback to default config why ever.

void IRsend::begin() {
	pinMode(IRpin, OUTPUT);
}

I'll look onto this...for now forgett the pull request, will close until I have really working code.

Sorry,

           Heiko.

closed 2017-02-11 by arendst