arendst/Tasmota #52: Home Assistant Gather Temperature Data via JSON

citrusfizz, 2017-02-10:
I'm trying to get Temp data with my DHT22 via HomeAssistant

Running this command

citrusfizz, 2017-02-10:
upon further inspection,  i think there is a "}" missing from

citrusfizz, 2017-02-10:
I have confirmed that the issue is the lack of a }  in the json ouput to MQTT

i haven't figured out where exactly to put this extra }  so i put it here

  if (dht_readTempHum(TEMP_CONVERSION, t, h)) {     // Read temperature
    dtostrf(t, 1, TEMP_RESOLUTION &3, stemp1);
    dtostrf(h, 1, HUMIDITY_RESOLUTION &3, stemp2);
    snprintf_P(svalue, ssvalue, PSTR("%s, \"DHT\":{\"Temperature\":\"%s\", \"Humidity\":\"%s\"}}"), svalue, stemp1, stemp2);

This sends 
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2017-02-10T21:16:38", "DHT":{"Temperature":"77.9", "Humidity":"22.3"}}} 
instead of 
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2017-02-10T21:16:38", "DHT":{"Temperature":"77.9", "Humidity":"22.3"}}
however, it is not correct as the normal telemetry data that is sent on an interval now has to many }

tele/sonoff/SENSOR {"Time":"2017-02-10T21:19:13", "DHT":{"Temperature":"73.0", "Humidity":"23.9"}}}

@arendst      I'm sure there is somewhere in the code to make sure the 
stat/sonoff/STATUS10  json response has the needed  } at the end.  but i have yet to figure out where exactly to put it. 

let me know if i need to clarify anything else.

arendst, 2017-02-10:
Thnx for the research! You're probably right. I will find the correct location for the brace and release an update next week.

citrusfizz, 2017-02-10:
No problem,   i did change on line in the code
 snprintf_P(svalue, ssvalue, PSTR("%s{\"Time\":\"%s\""), svalue, stime);
to  snprintf_P(svalue, ssvalue, PSTR("%s{\"Time\":\"%s\"}"), svalue, stime);

and that fixed the response for the stat/sonoff/STATUS10
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2017-02-10T22:20:55"}, "DHT":{"Temperature":"70.9", "Humidity":"27.2"}}
but it broke the  format for the periodic telemetry 
tele/sonoff/SENSOR {"Time":"2017-02-10T22:20:53"}, "DHT":{"Temperature":"70.9", "Humidity":"27.2"}}

so the first didn't add another } at the end,  it  inclosed the Time: field
which resulted in the periodic telemetry (which was working fine) to now be screwed up. 

I'm just kinda writing down what i find here.   hopefully it will make it easier for you to fix the code. 

cheers

Mickbaer, 2017-02-11:
Hello,
I have found the same Thing, that there is existing an Problem with the JSON Strings.

STATE {"Time":"2017-02-11T00:08:53", "Uptime":127, "POWER":"off", "Wifi":{"AP":1, "SSID":"Spandau5", "RSSI":74}} 

I have got this error message:
 MQTT_DEVICE sonoffpow1842: WARNING: deformed JSON data, check your config.
and
MQTT_DEVICE sonoffpow1842: malformed JSON string, neither array, object, number, string or atom, at character offset 0 (before "incoming publish rec...") 

This came up with all from the Sonoffxx sendings in MQTT JSON Format strings.

So I think citrusfizz  directed in the correct direction, that there is some Thing faulty at the beginning of the JSON String.
It may be the missing "}" braked, but I'm not sure.

@Theo, I hope you can solve it.

Thanks in advanced.
 Greatings from Mickbaer Berlin Germany

By the way, Theo, another question:
I have noticed you are using at each JSON Sting at the time stamp you are using:
as Example:
STATE {"Time":"2017-02-11T00:08:53", 
Waht is about the "T" between Date and Time?
From where you have such Format to be used?
I know only an BLANK between Date and Time.

What is the reason, you are using this special Format?

arendst, 2017-02-11:
@citrusfizz 
Fix is change in sonoff.ino line 1931:
```
    snprintf_P(svalue, sizeof(svalue), PSTR("%s}"), svalue);
```

@Mickbaer 
According to [JSONLint](http://jsonlint.com/) and me there is nothing wrong with
```
 {"Time":"2017-02-11T00:08:53", "Uptime":127, "POWER":"off", "Wifi":{"AP":1, "SSID":"Spandau5", "RSSI":74}}
```
Where do you get the error message from?

About the datetime string: As I (and most software) hate spaces there is a T between date and time string.

arendst, 2017-02-11:
@citrusfizz Fixed in version 3.9.14 just released

citrusfizz, 2017-02-11:
Awesome,  Quick turn around.  I will test it now

citrusfizz, 2017-02-11:
@arendst 
Works perfect now.  Thanks for fixing that. 
For anyone else looking to get this working with Home Assistant 
here is my config

sensor:
  - platform: mqtt
    state_topic: 'stat/sonoff/STATUS10'
    command_topic: 'cmnd/sonoff/status'
    payload: '10'
    name: 'Temperature'
    unit_of_measurement: '°F'
    value_template: '{{ value_json.StatusSNS.DHT.Temperature }}'

 - platform: mqtt
    state_topic: 'stat/sonoff/STATUS10'
    name: 'Humidity'
    unit_of_measurement: '%'
    value_template: '{{ value_json.StatusSNS.DHT.Humidity }}'

  - platform: mqtt
    state_topic: 'tele/sonoff/SENSOR'
    name: 'Temperature'
    unit_of_measurement: '°F'
    value_template: '{{ value_json.DHT.Temperature }}'
 

The first two  will pick up the temp and humidity on the DHT22 sensor output when there is a mqtt pub request like this
mosquitto_pub -h localhost  -t 'cmnd/sonoff/status' -m '10'

This will be called from the HASS application, i just haven't gotten to that part yet.  so if you load this exact same config (as long as you have your MQTT server settings defined in HASS), it will read data when that mosquitto pub is ran. 

the last sensor picks up what the SONOFF and this software sends periodically and it done automatically at this point.   This periodic interval can be changed (see the wiki or the MQTT commands)

ok i am two hours late for a family function,   gotta run. 

Thanks again @arendst

arendst, 2017-02-11:
Thnx. I'll add your config to the wiki as many people seem to have trouble configuring Home Assistant

closed 2017-02-11 by arendst