arendst/Tasmota #55: Crash when select "Sonoff 4CH"

Reinhartd, 2017-02-11:
when i select "Sonoff 4CH" the System is rebooting and can no more starting!
I must reflashed the Module!

LG

arendst, 2017-02-11:
What version are you using?

If you just upgraded from what version did you came?

When you updated was the sonoff selected as Sonoff 4CH?

BTW: my wemos works fine as Sonoff 4CH albeit with one relay...

Reinhartd, 2017-02-11:
i have tested with Tasmota 3.9.12 + 3.9.13 and the Sketch before was MQTT 3.1.?.
When i flash MQTT 3.2.13, this  is ok.

LG

arendst, 2017-02-13:
Still waiting for mine to arrive. I'll test it then.

Reinhartd, 2017-02-13:
ok, Thank you!
i have still tested the version Tasmota 3.9.14, but the same Problem, reboot and can no starting. No Led is blinking and can not see an AP or ping the old ip.

arendst, 2017-02-14:
As the 4ch uses the same chip as the touch some special handling regarding programming is needed.

I suppose you fallback scenarion is recompiling for esp8285 and serial upload the code. That procedure should also work for 3.9.15

Ota or file upload from 3.2.13 to any other version will always fail unless the image is compiled for esp8285 which uses flash mode dout.

Starting with version 3.9.12 I check if the module is set for touch or 4ch and if so patch the uploaded image for flash mode dout before it is copied to it's final destination.

If notto much trouble could you recompile 3.9.15 for esp8285 and serial upload to 4ch.After restart set module to 4ch. From now on esp8266 compiled images should upload just fine.

Thnx in advance as I'm still waiting for mine to arrive...

Reinhartd, 2017-02-15:
Thanks, 

i have compiled Version 3.9.16 with Arduino and settings "8285" and preselect "Sonoff_4CH" and upload serial to 4CH!
Now it is all ok, the Module can start!

LG

arendst, 2017-02-15:
Wiki updated with Tips fo both Sonoff Touch and Sonoff 4CH regarding initial firmware load and future OTA and Upload upgrades.

closed 2017-02-15 by arendst