arendst/Tasmota #57: Extra GPIO

citrusfizz, 2017-02-12:
I noticed that i can set GPIO3 on my normal sonoff  and use the RX pin  as a GPIO and then of course we have GPIO14 at the end  which we can use.  But where is GPIO4?  i tested the TX pin but no luck. 

Does the normal sonoff have GPIO 4 to use somewhere? or did i just get really lucky with gpio3 working with the RX pin on the sonoff?

arendst, 2017-02-12:
Only Sonoff SV has an easy accessable GPIO04 available. Sonoff TH can deliver GPIO04 on it's socket after soldering some jumpers to the board.

People seem to use the software on wemos too where most pins are available...

citrusfizz, 2017-02-12:
thanks for the succinct answer.    Interesting that people use this on wemo devices tho, they aren't esp based are they?  from what i can find  they use a Ralink chip

jaysuk, 2017-02-12:
The wemos d1 is an ESP8266 based device

closed 2017-02-12 by arendst

citrusfizz, 2017-02-12:
interesting.. thanks for the answers,  you can close this issue if you want @arendst