arendst/Tasmota #58: adding support for latching relay

Micha001, 2017-02-12:
Hello, 

I've added the wall switch device from the link below:
[https://ex-store.de/ESP8266-WiFi-Relay-V31]

It uses a latching type relay instead of a normal one. Therefore it needs a secondary output pin to reset the relay. 

It would be great if you could add the code snippets to your repository. Maybe someone will use it and for me it becomes easier to follow your updates in future.

file sonoff.ino:

global variable:
-----------------------------------------------------------------------------
uint16_t mqtt_cmnd_publish = 0;       // ignore flag for publish command
.
.
 **uint8_t relay_pulse = 0;**
.
.
#ifdef USE_MQTT_TLS

function void do_cmnd_power(): 
-----------------------------------------------------------------------------
   power ^= mask;
    }
.
.
		   **if (sysCfg.module == EXS_RELAY){
				if (!(power & 0x01)){ 
					power |= 0x02;
					}
					else power &= (0xFF ^ 0x02);
			   if (relay_pulse) power &= (0xFF ^ 0x03);    
			   relay_pulse = 2;
			   }**    
.     
.    
      setRelay(power);
	  
function void stateloop():	  
-----------------------------------------------------------------------------

  if (!pulse_timer) do_cmnd_power(1, 0);
  }
.
.
		  **if (sysCfg.module == EXS_RELAY && relay_pulse){   
			relay_pulse--;
			if (!relay_pulse)setRelay(power & (0xFF ^ 0x03)); 
		  }**
.
.
  if (blink_mask) {
.
.
.
 power_now = (power & (0xFF ^ blink_mask)) | ((blink_power) ? blink_mask : 0);
.
.
		 **if (sysCfg.module == EXS_RELAY){
				  if(!(power_now & 0x01)){power_now |= 0x02;
				  }
				  else power_now &= (0xFF ^ 0x02);
				   relay_pulse = 2;
				}**
.
. 
 setRelay(power_now);
  
 file sonoff_template.h:
--------------------------------------------------------------------------- 

  ELECTRODRAGON,
  USER_TEST,
.
. 
			**EXS_RELAY,**
.
.
  MAXMODULE }; 
. 
. 
   0, 0
  },
 .
 .
		  **{ "EXS Relay",           // latching relay https://ex-store.de/ESP8266-WiFi-Relay-V31   
			 GPIO_USER,          // GPIO0 -Modul Pin 8
			 GPIO_USER,          // GPIO01 UART0_TXD; GPIO1 -Modul Pin 2  
			 GPIO_USER,          // GPIO02 -Modul Pin 7
			 GPIO_USER,          // GPIO03 Serial RXD and Optional sensor -Modul Pin 3 
			 GPIO_USER,          // GPIO04 Optional sensor -Modul Pin 10 
			 GPIO_USER,          // GPIO05 Optional sensor -Modul Pin 9 
			 0, 0, 0, 0, 0, 0,  
			 GPIO_REL2,          // GPIO12 Relay1 ( 1 = Off) 
			 GPIO_REL1,          // GPIO13 Relay1 ( 1 = on)  
			 GPIO_USER,          // GPIO14 Optional sensor -Modul Pin 5
			 0,                  // GPIO15 not used 
			 GPIO_USER           // GPIO16 Optional sensor -Modul Pin 4 
		  }                      // -Modul Pin 1 VCC 3V3; -Modul Pin 6 GND**  
.
.

altelch, 2017-02-12:
Great, was looking for a wall socket mounted device.

arendst, 2017-02-12:
Implemented in next release

arendst, 2017-02-13:
Adjusted and implemented in version 3.9.15 just released. Give it a go and let me know if my changes did not break your functionality. I had to test using two leds....

Micha001, 2017-02-18:
Great! :+1: 

I've tested it, looks good. 

Thanks to you

closed 2017-02-24 by arendst

sfromis, 2022-10-06:
Did you try if the recently (version 12.1) added Tasmota support for bistable/latching relays is suitable? You can select the pulse length with `SetOption45`.

rarescirca, 2022-10-06:
Nope I dont use it, thanks. 
I just Upgrade the Firmware and the relay_b looks oke, but only one output is working.
Should I group the relay GPIO somehow that Tasmota know is the same relay?
Attached is the current configuration.
In the second photo you will se the output only for one GPIO.
The Set pulse option is perfect :)

Rares

[image]
[image]

sfromis, 2022-10-06:
Same channel number (dropdown to the right) cannot work, each occurrence of a component needs a separate number. I'm not aware of an option to suggest that it is really the same relay, but not having worked with bistable relays, I have no conclusive info on how to deal with this.

arendst, 2022-10-06:
You need a gpio pair for every bistable relay called relay_b1 and relay_b2

rarescirca, 2022-10-07:
I can confirm that @arendst solutions works,
Thanks guys, much appreciated.
 
[image]

arendst, 2022-10-17:
Latest tasmota supports a number of bilatching relays. Every supported bilatching relay needs to have a GND or Vcc connection and two coils (mostly using driver transistors) connected to two gpios configured in a pair as ``Relay_b1`` and ``Relayb_2``. The next relay would be connected to GPIOs called ``Relay_b3`` and ``Relay_b4``.

The default pulse length is set to 40 mS but can be changed online with command ``SetOption45``

ahmaddxb, 2023-04-09:
I'm testing this with 4 led's connected to each relay_b1-4. I expect when I toggle 1 on led 1 will pulse and when i toggle 1 off led 2 will pulse. Same goes for toggle 2 on led 3 pulse / off led 4 pulse.

This is no the behavior I'm experiencing. 
Toggle 1 on = led 1 and 4 pulse
toggle 1 off = led 2 and 4 pulse
Toggle 2 on = led 3 pulse
toggle 2 off = led 2 and 4 pulse

Is this behavior I'm experiencing they way it should be?

Use case is to use a h bridge motor controller to have generate positive pulse from relay_b1 and then a negative pulse from relay_b2, this does work as expected but then when adding relay_b3/4 things get messed up.

[image]

arendst, 2023-04-09:
Your expectation is wrong.

On a "normal" relay power switch every configured relay is set to the state of the 32-bit power register to make sure Tasmota knows the state of every relay.

The same is true for bi-stable relays; every bistable relay is set to the position the 32-bit power register holds. So what you see exactly expresses the designed functionality.

So do not mis-use bi-stable relays control for H-bridge control.

ahmaddxb, 2023-04-09:
so what is the best option, if i set them as normal relay then i would have two buttons used to control the on and the off functions.

I want to control a solenoid that requires a positive pulse to open and then a negative pulse to close. the h bridge motor controller can do that and requires 2 inputs.

barbudor, 2023-04-09:
@ahmaddxb you can use normal relays with a pulsetime to insure only a brief on pulse will occur. If you need to maintain a state of the solenoid the you can use rules to set a variable accordingly.
I admit that will not give you a virtual relay that represent the state of the solenoid.

ahmaddxb, 2023-04-09:
Yes i did use pulse time before. I was hoping for a way to keep it all in tasmota but I guess I can make a single virtual switch in home assistant that can toggle each relay. Thanks both of you.

barbudor, 2023-04-09:
@ahmaddxb
Looking again on how bi-stable relays works, I don't see that a problem with your solenoid
What it does is repeating all states for all Power
so when toggling Power1 (=Relay_b1+Relay_b2) it also repeat the value of Power2 (if Power2 is off, it will pulse Relay_b3, if it's one it will pulse Relay_b4)
Which means that this Power2 (or your 2nd solenoid in your case) will just receive a pulse for the position it is already there

So it shouldn't be a problem

ahmaddxb, 2023-04-09:
@barbudor
Yes, I already came to that conclusion after i thought about it and understood what was going on. Testing on the breadboard and seeing other led's start flashing just throw me off, I didnt know about the 32-bit power register. I assumed it was weird behavior and might cause me issues.