arendst/Tasmota #60: Password protection for web interface?

smadds, 2017-02-12:
I would like to discuss the feasibility and appetite for adding password protection to the web interface.
It seems to me that at the moment the Sonoffs are vulnerable to anyone logged into the same wifi network. Clearly one option is to create a hidden SSID wifi access point for Sonoff use, but if that is not an option, how can we secure the devices?

Would an admin password setting option on the 192.168.4.1 initial setup page work? If the password was forgotten a full system reset could start from scratch. Also, setting/resetting the admin pw from MQTT could be possible as access to the MQTT server can be secured.

Thoughts?

arendst, 2017-02-13:
Quick solution: disable webserver with option 0.

smadds, 2017-02-13:
If there was a "like" button I would press it!
ᐧ

hoekbrwr, 2017-02-13:
/play bezos

hoekbrwr, 2017-02-13:
:closed_lock_with_key:

closed 2017-02-15 by arendst