arendst/Tasmota #61: Compile error

0n3-70uch, 2017-02-14:
Hello Theo,

I want to compile the new version 3.9.16. But this error apperars:

> /home/fabian/Entwicklung/Arduino/Sonoff/Sonoff.ino:158:27: fatal error: ArduinoJson.h: No such file or directory
   #include <ArduinoJson.h>
                           ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module. (Error while compiling the code for the Board (...)).

Greetings
Fabi

altelch, 2017-02-14:
You need to install ArduinoJson library from library manager.

Greetings,
Heiko.

0n3-70uch, 2017-02-14:
Hello Heiko,

yeah now I notice that there are some more libaries are required. The next step is the IRremoteESP8266.h.
I was a little bit to fast... the wiki is not up to date because the code is to new :-).

Greetings
Fabi

closed 2017-02-14 by 0n3-70uch

0n3-70uch, 2017-02-14:
I updated the wiki page:
https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite