arendst/Tasmota #62: Same number for GPIO configuration after update

0n3-70uch, 2017-02-14:
Hello,

just a short thing:
I think its better to have the same GPIO type numbers for every firmware version. With the last update the module "Switch" changed from 8 to 9. I only confiure my modules with MQTT commands. So first I was a little confused, because the switch was not working. Than I notice that the number changed to 9. I think its better to add new functions at the end. 

For example (last update):
8 > Switch
9 > IRremote

What do you think about that?
Greetings
Fabi

arendst, 2017-02-14:
I know the consequences but as more swithes may be made available someday it needs to be the last one in the list.

Sorry for inconveniences this may have.

0n3-70uch, 2017-02-14:
Haha no problem. Now I understand your way...

An idea: make some categories. Like this...
10 temperature Sensor DHT
11 temperature Sensor AMS
20 switch type 1
21 switch type 2
...
29 switch type 10
30 led WS2812
31 led test

closed 2017-02-15 by arendst