arendst/Tasmota #65: Touch lost after selecting Module type

yllar, 2017-02-14:
Did OTA from 3.2.13 to 3.9.16  and the Touch booted up fine, selected Module type 10 and it didn't start again.
log from mqtt:
stat/touch-1/RESULT {"OtaUrl":"http://192.168.1.200/ota/firmware.bin"}
stat/touch-1/RESULT {"Upgrade":"Version 3.2.13 from http://192.168.1.200/ota/firmware.bin"}
tele/touch-1/LWT Offline
tele/touch-1/LWT Online
stat/touch-1/RESULT {"POWER":"OFF"}
stat/touch-1/POWER OFF
stat/touch-1/UPGRADE Successful. Restarting
tele/touch-1/LWT Offline
tele/touch-1/LWT Online
cmnd/touch-1/POWER (null)
stat/touch-1/RESULT {"POWER":"OFF"}
stat/touch-1/POWER OFF
tele/touch-1/INFO1 {"Module":"Sonoff Basic", "Version":"3.9.16", "FallbackTopic":"DVES_82ED4A", "GroupTopic":"sonoff2s"}
tele/touch-1/INFO2 {"WebserverMode":"Admin", "Hostname":"touch-1-3402", "IPaddress":"192.168.1.148"}
stat/touch-1/RESULT {"POWER":"OFF"}
stat/touch-1/POWER OFF
tele/touch-1/STATE {"Time":"2017-02-14T20:52:17", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"NSABuss1", "RSSI":98}}
tele/touch-1/LWT Offline

PS. OTA and config successful with S20 and POW, great job!

arendst, 2017-02-14:
Touch uses esp8285 with dout flash programming. From your log I see you set touch to basic module and then ota update. Doing it that way does not change the flash flag from dio to dout and next boots will fail.

Correct steps now are
- recompile for esp8285 orset flash flag to dout and recompile
- serial upload new code
- select Sonoff Touch and you are good to go for future ota uploads as now the flash flag will be corrected before the code is cpied to it's final destination.

Glad the others went fine.

yllar, 2017-02-14:
I cannot make it to talk to IDE anymore and it looks like the magic smoke has escaped :(
[image]
What i did:
1. downloaded https://github.com/arendst/Sonoff-Tasmota/releases/download/v3.9.16/firmware.bin
2. uploaded it via web OTA url
3. Touch rebooted and started as basic
4. selected Sonoff Touch as module
5. it rebooted and never came back 

Should the firmware file work with Touch if it's esp8285 based unlike others?

arendst, 2017-02-14:
Well it did for a short moment... Anyway it should not blow the hardware.

See #55 for an explanation regarding esp8285

yllar, 2017-02-14:
Board: "GenericESP8285 Module"
Flash Mode: "DOUT"
CPU Frequency: "80 MHz"
Flash Size: "1M (64K SPIFFS)"
Upload Speed: "115200"

GPIO0 is connected to ground 
Upload fails with:
```
Sketch uses 460976 bytes (48%) of program storage space. Maximum is 958448 bytes.
Global variables use 44944 bytes (54%) of dynamic memory, leaving 36976 bytes for local variables. Maximum is 81920 bytes.
error: setattr stage 1 failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```
I was able to flash Basic, so FTDI is fine
Guess i'll better order new as these things take ages to arrive :D

arendst, 2017-02-15:
Wiki updated with Tips regarding Sonoff Touch and Sonoff 4CH firmware upgrades.

closed 2017-02-15 by arendst