arendst/Tasmota #68: Use Sonoff as momentary relay

SecurityAddicted, 2017-02-15:
Hi, just a question, can the Sonoff be used as momentary relay with the TASMOTA firmware?
I mean, activate, wait a specific amount of time, then deactivate, sending it a specific command?
Thanks.

Nayar, 2017-02-15:
I guess you'll have to use a third party software run a timer and send the required commands to activate and deactivate. Try Homeassistant

jaysuk, 2017-02-15:
Or node red

arendst, 2017-02-15:
Why not using pulsetime? Set it to a value like 100 and after a power on command it turns off after 10 seconds.

See the wiki for more functional commands...

SecurityAddicted, 2017-02-15:
Thanks to everyone for the tips :)
Theo, just saw the pulsetime option in the wiki... there is any documentation about the commands around?

davidelang, 2017-02-19:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

I missed it as well

closed 2017-03-08 by arendst