arendst/Tasmota #69: sonoff pow keeps forgetting today value

octek0815, 2017-02-15:
Hi,

my POW keeps forgetting the "Today" value.

**before ac remove**
19:00:47 RSLT: tele/sonoff/STATE = {"Time":"2017-02-14T19:00:47", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"N-O-D-D-E-S", "RSSI":84}}
19:00:47 RSLT: tele/sonoff/ENERGY = {"Time":"2017-02-14T19:00:47", "Yesterday":0.000, **"Today":0.012**, "Period":0, "Power":54, "Factor":1.00, "Voltage":214, "Current":0.244}
19:00:58 MQTT: Attempting connection...
19:00:59 mDNS: Query done with 0 mqtt services found
19:01:09 RSLT: tele/sonoff/STATE = {"Time":"2017-02-14T19:01:09", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"N-O-D-D-E-S", "RSSI":86}}
19:01:09 RSLT: tele/sonoff/ENERGY = {"Time":"2017-02-14T19:01:09", "Yesterday":0.000, "Today":0.012, "Period":0, "Power":54, "Factor":1.00, "Voltage":214, "Current":0.246}
19:01:16 MQTT: Attempting connection...
19:01:17 mDNS: Query done with 0 mqtt services found

**after power restore**
19:01:51 mDNS: Query done with 0 mqtt services found
19:02:03 RSLT: tele/sonoff/STATE = {"Time":"2017-02-14T19:02:03", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"N-O-D-D-E-S", "RSSI":82}}
19:02:03 RSLT: tele/sonoff/ENERGY = {"Time":"2017-02-14T19:02:03", "Yesterday":0.000, **"Today":0.007**, "Period":0, "Power":54, "Factor":1.00, "Voltage":213, "Current":0.245}
19:02:08 MQTT: Attempting connection...
19:02:09 mDNS: Query done with 0 mqtt services found

**wait for Today > 0.007...**
19:02:45 mDNS: Query done with 0 mqtt services found
19:02:54 RSLT: tele/sonoff/STATE = {"Time":"2017-02-14T19:02:54", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"N-O-D-D-E-S", "RSSI":84}}
19:02:54 RSLT: tele/sonoff/ENERGY = {"Time":"2017-02-14T19:02:54", "Yesterday":0.000, **"Today":0.008**, "Period":0, "Power":54, "Factor":1.00, "Voltage":212, "Current":0.246}
19:03:02 MQTT: Attempting connection...
19:03:03 mDNS: Query done with 0 mqtt services found

**with Reset 2 from commds it remebers last Today value.**
19:04:58 mDNS: Query done with 0 mqtt services found
19:05:11 RSLT: tele/sonoff/STATE = {"Time":"2017-02-14T19:05:11", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"N-O-D-D-E-S", "RSSI":82}}
19:05:11 RSLT: tele/sonoff/ENERGY = {"Time":"2017-02-14T19:05:11", "Yesterday":0.000, **"Today":0.009**, "Period":0, "Power":55, "Factor":1.00, "Voltage":216, "Current":0.246}

I realy don´t understand why this ist happening. I have two pow´s and both have the same problem.

Regards,
Oliver

arendst, 2017-02-15:
Today power is dynamic. Saving it's value to flash on every change wil blow your hardware within months. So today power is saved only at 00:00 and is then called yesterday.

You should save the values yourself using a home automation tool like domoticz.

octek0815, 2017-02-16:
Ok, thx for your quick reply. Then i will try to solve it in FHEM.

closed 2017-03-08 by arendst