arendst/Tasmota #71: Home assistant MQTT support

aequitas, 2017-02-15:
I'm looking into supporting the WS2812 commands in home assistant but the current HA components (light.mqtt/light.mqtt_template) don't allow this easily yet. I'm considering some refactoring or a new component for HA to allow better integration (https://github.com/home-assistant/home-assistant/pull/5879).

Another alternative I'm considering is some addition to the Tasmota MQTT protocol. I have not done an investigation on the Tasmota MQTT code, but I'm thinking somewhere in the line of one MQTT topic to which multiple commands can be send at once `command=value` style or maybe JSON. This way integration could be easily made using a template like: https://home-assistant.io/components/light.mqtt_template/#json-payload

Whats your opinion on this, would it be sensible or would it add to much complexity to the code/protocol?

altelch, 2017-02-15:
Hi,

this is partially implemented in the HUE emulation part. I plan to support more HUE features in future. On the other hand after adding ArduinoJSON to the built requirements parsing JSON MQTT isn't to difficult now. MQTT handling for WS2812 is done in xdrv_ws2812.ino in ws2812_commmand. An example for parsing JSON is in xdrv_ir_send.ino.

Greetings,
    Heiko.

aequitas, 2017-02-15:
Thanks for the leads. I will look into if the HUE emulation could work as a solution in this case.

closed 2017-03-08 by arendst