arendst/Tasmota #72: First firmware compilation with PlatformIO not working for Sonoff 4CH

lucwuyts, 2017-02-16:
I'm a first time user of this project !! 
I like it very much.

I wanted to use this on a Sonoff 4CH 
With the setting for ESP8285, compilation is not possible because PlatformIO thinks the size of the firmware is too big for this chip.

With the Arduino IDE, there was no problem at all.

Maybe you should add this to the wiki on the prerequisites page.

arendst, 2017-02-16:
I see no problem if I compile it with the default user_config.h and platformio.ini settings:
```
[platformio]
src_dir = sonoff

[env:sonoff]
platform = espressif8266
framework = arduino
; ESP8266 - Sonoff Basic et al.
; board = esp01_1m
; ESP8285 - Sonoff Touch and Sonoff 4CH
board = esp8285
build_flags = -Wl,-Tesp8266.flash.1m64.ld -DMQTT_MAX_PACKET_SIZE=400

lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
```

lucwuyts, 2017-02-16:
Hi Theo,
This is what i get:

Compiling .pioenvs\sonoff\src\sonoff.ino.o
Linking .pioenvs\sonoff\firmware.elf
Checking program size
text       data     bss     dec     hex filename
Error: The program size (455211 bytes) is greater than maximum allowed (434160 bytes)
*** [.pioenvs\sonoff\firmware.elf] Explicit exit, status 1
444611    10600   35056  490267   77b1b .pioenvs\sonoff\firmware.elf
 [ERROR] Took 7.48 seconds

arendst, 2017-02-16:
Strange. Are you sure you are compiling with the platformio.ini as shown previously? It looks like it compiles for a 512k memory image while it should compile for 1M image...

I'm running with platformIO versions IDE 1.7.1 and Core 3.2.1

lucwuyts, 2017-02-16:
I copy and pasted the configuration from your earlier post, but still the same.

This is with platformIO IDE 1.7.2  and Core 3.2.1

arendst, 2017-02-16:
Updated PlatformIO to your versions and still works fine:
```
Linking .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Calculating size .pioenvs\sonoff\firmware.elf
text       data     bss     dec     hex filename
452371    10836   35080  498287   79a6f .pioenvs\sonoff\firmware.elf
 [SUCCESS] Took 12.74 seconds
```

lucwuyts, 2017-02-16:
I removed platformIO and the datafolder, reinstalled, but still the same problem.

The ArduinoIDE is working, so i will keep using that in the future.

jonathanmbradshaw, 2017-02-23:
Same here, running the build from the TAR file on 64bit Fedora 
PlatformIO @ IDE 1.7.2 | Core 3.2.1

platformio run --target upload
text       data     bss     dec     hex filename
455169     9920   34600  499689   79fe9 .pioenvs/sonoff/firmware.elf
Error: The program size (465089 bytes) is greater than maximum allowed (434160 bytes)

LesFragnes, 2017-02-26:
Same memory problem here on win8, PlatformIO @ IDE 1.7.1 | Core 3.2.1, Sonoff Touch EU Power V1.0 

As a work around I commented out anything I didn't need in user_config.h to save memory until it compiled and loaded without error.

Many thanks to Arendst for the brilliant firmware.

arendst, 2017-03-08:
The memory definition in platformio for esp8285 is wrong.

Today I ve update the platformio.ini file with better parameters. Give it a go.

closed 2017-03-12 by arendst

jonathanmbradshaw, 2017-04-01:
Sorry for the delay, looks like its working, thanks and all the best Jonathan

mrcsmxms, 2017-07-21:
i downloaded the code a few days ago and still have the same problem:

this is my setting:
; Sonoff et al (ESP8266 based)
[env:sonoff]
platform = espressif8266
;platform = ESP8285
framework = arduino
; board = esp01_1m
board = esp8285
board_flash_mode = qio
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

and then i get the following error:

Error: The program size (478949 bytes) is greater than maximum allowed (434160 bytes)

davidelang, 2017-07-21:
On Fri, 21 Jul 2017, Marcus wrote:

> Error: The program size (478949 bytes) is greater than maximum allowed (434160 bytes)

you need to set the flash size to 1M (no spiffs)