arendst/Tasmota #76: Connot compile

yann25, 2017-02-17:
Hello, I'm new here and I have a question concerning the flashing of an ESP8266 described in in this documentation.

 I have tried PlatformIO and the Arduino IDE. 
I followed every step in the tutorial, deleted everything after it did not worked and redid every step. :see_no_evil:

PlatformIO says: 

>     File "c:\.pioidepenv\lib\site-packages\platformio\managers\package.py", line 291, in download
>         fd = FileDownloader(url, dest_dir)
>       File "c:\.pioidepenv\lib\site-packages\platformio\downloader.py", line 51, in __init__
>         self.set_destination(join(dest_dir, self._fname))
>       File "c:\.pioidepenv\lib\ntpath.py", line 85, in join
>         result_path = result_path + p_path
>     UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 13: ordinal not in range(128)

and some more before.

Arduino IDE says that Header files are missing.

Since this is the first time I used PlatformIO I don't know how to go on. With the Arduino IDE I tried diffrent versions and also copied the header files from elsewhere but it did not work. I have no idea what to do next so I created an accout to ask the community :D

Thanks for your help :)

hoekbrwr, 2017-02-18:
It was no problem for me using Arduino IDE, it still works. Just grab the sonoff.ino in the sonoff directory and follow the wiki. You only have to adapt the name of the serial port you are trying to use. You can check in Device Monitor under Ports if there are any problems with drivers for your serial connection hardware.

yann25, 2017-02-20:
Thank you for your answer, I will try it again later. I had no problem flashing it with another software, so I thought there was something missing, but if you say it worked I might have made a mistake.

yann25, 2017-02-20:
After digging in the PlatformIO GitHub page I found out that the PlatformIO parses through the whole project to find dependencies. For this Python is used with utf-8 I think. My username has an "ö" in it so in couldn't parse the project. 
That was the problem...

closed 2017-02-20 by yann25