arendst/Tasmota #77: Reset when use serial port monitor

Greg2002, 2017-02-17:
3.9.17 20170217
1.When use a serial port - device resetting:
(Config - Sonoff, GPIO4+GPIO14 as I2C), below - a log
00:04:42 MQTT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:04:43 Config: Use default configuration
00:04:57 SYSL: Syslog Host not found so logging disabled for 600 seconds. Consider syslog 0
00:04:58 APP: Restarting....
and so on loop
2. When you switch On Sonoff -the very first second relay is On (!), after loading - it become On or Off as in last stage. It's dangerous.

hoekbrwr, 2017-02-18:
You mean Web Console? If you try to use serial monitor with pin 4 using I2c then there is a conflict because pin 4 is normally used for serial communication.

Greg2002, 2017-02-18:
No, not Web Console. I tried to read information from Arduino IDE serial port. After that Sonoff resetting to default configuration.

hoekbrwr, 2017-02-18:
All the things you do on the HW serial connection can be done from the Web Console. You cannot use a serial connection when defining pin 4 as I2c. The FW will produce a reset as you experienced. So search for the IP address of your Sonoff on your local network and type <IP-address> in the address line of you favorite browser. Look for commands in the wiki side menu
!

Greg2002, 2017-02-18:
OK. I know it. It's not necessary read data from Arduino IDE when you have Console and MQTT.
I think that only RX TX pins could conflict with serial port. Thank you

closed 2017-02-22 by arendst