arendst/Tasmota #78: Dirive WS2812B 5 Meter Stripe with Sonoff Basic

thommycgn, 2017-02-18:
Hi, 

I just bought a WS2812B 5Meter 300 LEDs Light Stripe and tried to make it run with Sonoff Basic. 
I tried to uncomment #define USE_WS2812_DMA  and use it on the RX (GPIO3) and even on the GPIO 14 but cann't really make it run. I connected 5V from the transformer directly to the stripes and the Data Pin with the GPIO of the Sonoff Basic. 
Only the first few LEDs are reacting and even not correctly. I tried it with RGB and GRB Mode. 
Did anybody get it run already?

altelch, 2017-02-18:
Hi,

did you connect Sonoff ground with power supply ground of the LED strip? This sounds like no common ground what causes a floating data pin with undefined results.

Greeting,

     Heiko.

thommycgn, 2017-02-18:
Hi, 

I connected 230V -> Sonoff Basic -> Trafo 
Then:  Trafo 5V Output -> WS2812B Stripe

The Datapin of the Stripes are connected to GPIO of Sonoff. 

Is it correct like that?

altelch, 2017-02-18:
Hi,

you are missing one connection. You supply the 5V from Trafo and Data from Sonoff. And I'd guess you connected the Trafo GND to Stripe GND but you must also connect Sonoff GND to Stripe GND. A common ground is important otherwise the GPIO Output has an undefined signal level for the strip.

Greetings, 

    Heiko.

hoekbrwr, 2017-02-18:
I do not really know exactly how much those 300 LEDs will try to suck out of that little trafo, but alltogether it could be 4A or more. So you need a separate supply and connect both grounds of Sonoff and separate supply together. As I remember there was about 1A for 60LEDs when full intensity white. Try and find the datasheet of WS2812b with Google.

altelch, 2017-02-18:
Hi, a WS2812b take max. 60mA per LED -> 300 LEDs are 18A what requires a good PSU.

Greetings,
      Heiko.

hoekbrwr, 2017-02-18:
So I have to check my Leds again. Probably not the same as yours. Mines are in a ring for a digital clock and other purposes, not yet defined.

closed 2017-03-08 by arendst

andreaslink-de, 2018-07-28:
Hey there, I have exactly the same setup here - even though this one is closed, this is exactly describing my need and my issues, so I give it a try anyway.

The only difference in my setup is that I already started with a common ground. So in detail, I have a Sonoff Basic with GPIO 14 connected to the data pin (including 220 Ohm) of the WS2812B stripe. GND from Sonoff is connected to GND from power supply (which is a 20A one) and 230V first enter the Sonoff, and leave it again connecting and serving the power supply. So the idea is, the relais of the Sonoff will start up the power supply which is the driver of the WS2812B LEDs on the stripe.

But before connecting the stripe, I was just testing the wiring with the external additional power supply for the stripe and this fails in a weird way. I boot up and everything is connected as described. But the moment I activate the relais via web the Sonoff disconnects and the green light stays on (WiFi aka ping stops). Now I manually turn the relais off via the button and the module seems to reboot or at least reconnect. 

So to do some analysis, I simply just removed the connected grounds and from that moment on, there is no problem anymore - but of course, then my common ground is missing. So something is wrong with the potentials here, I guess, I cannot drive the setup with a common ground because that causes the Sonoff to fail/die. But I have no clue, what's wrong here?
Did anyone solve this riddle or has a comparable setup successfully up an running?

**What I want** is to use the Sonoff Basic to switch the powerful power supply on for the WS2812B stripe and with the unused GPIO 14 I'd like to control the LEDs on that stripe powered by the just turned on power supply. 
So this should be a fairly common approach, isn't it?

captFuture, 2019-02-19:
Good Morning,

My setup is exactly the same as andreas and i have the same issue.
After the first Power ON of the psu via the sonoff relais and activating the ledstrip on GPIO14 the device gets completely unresponsive and does not connect to wifi any more.

joba-1, 2019-02-19:
Since Andreas mentioned common approach: I don't know what the issue is, but I wouldn't do it this way.

Use an additional NodeMCU (just +2€) that is powered by the 5V power supply to control the strip.
Much safer to physically separate 230V from the 5V strip, and you can use other firmware for controlling the strip with much more features than what tasmota currently can give you. Works well for me.

captFuture, 2019-02-19:
> Since Andreas mentioned common approach: I don't know what the issue is, but I wouldn't do it this way.
>
> Use an additional NodeMCU (just +2€) that is powered by the 5V power supply to control the [...]

Thanks a lot, that's what I just started to do today. Sonoff for Power on/off and an ESP32 for controlling the leds.
(another solution would be possible as well, like using a Wemos D1 With Tasmota as additional light controller which interacts with the first device ... if somebody wants to stick to the "tasmota-way")
:D