arendst/Tasmota #79: MQTT connection not always working

lucwuyts, 2017-02-18:
When i restart the sonoff, the MQTT connection is immediately ok.

But after a few seconds i often see these messages:  

MQTT: Attempting connection...
MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds

Return code 2 is Connection Refused, identifier rejected

during this experiment there was only ESP device active

Is this a configuration issue on the broker ?

lucwuyts, 2017-02-18:
I recompiled with version 3.9.17 and used the same user_config.h  and it is working as it should!

lucwuyts, 2017-02-20:
Today, I've seen this errors again.
My MQTT broker is mosquitto (on windows)

Error messages on the sonoff device:
14:56:23 MQTT: Attempting connection...
14:56:29 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds

Firmware is not changed.

lucwuyts, 2017-02-20:
I think the keep-alive pings are sometimes not sent

This is what i see on the broker side

 Client DVES_82579D has exceeded timeout, disconnecting.
 Socket error on client DVES_82579D, disconnecting.

lucwuyts, 2017-02-20:
And when the connection is accepted on the broker, the sonoff dos not always respond

1487600218: New connection from 10.0.250.72 on port 1883.
1487600218: New client connected from 10.0.250.72 as DVES_82579D (c1, k15, u'mqtt-user').
1487600218: Sending CONNACK to DVES_82579D (0, 0)
1487600239: Client DVES_82579D has exceeded timeout, disconnecting.
1487600239: Socket error on client DVES_82579D, disconnecting.

arendst, 2017-02-21:
Still need to investigate but did you make any changes to pubsubclient.h define MQTT_SOCKET_TIMEOUT?

I used it as set to 2 seconds for a while to satisfy non MQTT users but I will now set it back to 15.

Can you verify your compiles did use the 15 second value and still observe the problem you see?

lucwuyts, 2017-02-21:
The library defines it to 15.
And according to the mosquitto-verbose output, it is indeed every 15 seconds.

This problem is not constant, and i begin to think actions in the web interface of the sonoff sometimes are causing this.

arendst, 2017-02-24:
I think it is webserver related too. Did a rewrite of the page refresh using ajax. Let's see if it was for the good.

Implemented in version 3.9.21 just released.

lucwuyts, 2017-02-25:
Yes, I think it is better now.
Tried a few times:  reboot and have the web-console screen open.
After a few attempts to connect (with the same error message as before), it now succeeds, and looks (for now) a stable connection.

closed 2017-03-09 by arendst

Sunrise17, 2018-01-04:
Is it possible to increase MQTT_KEEPALIVE from 15secs to 60secs from Web-Console with MQTT command?

webtop, 2018-01-04:
As a developer, if you need a socket timeout of 60 seconds, the issue is not on the device firmware, but on the server socket.
Where is your MQTT broker running?

Sunrise17, 2018-01-04:
i need to increase MQTT_KEEPALIVE to 60 secs, which is arranged 15secs in Sonoff Basic Tasmota FW?

Sunrise17, 2018-01-04:
My MQTT broker (on Pi) MQTT_KEEPALIVE's setting is already 60secs

webtop, 2018-01-04:
Right, my point exactly. If you have the timeout maxed out and you are on the same network with your MQTT broker, the issue is not in the FW, the issue is more than likely the gain on your WiFi (check RSSI levels) or in access-control on the Pi (make sure anonymous connection available to MQTT to test)

webtop, 2018-01-04:
@Sunrise17  Please open your own thread so it can be discussed in there.

careyer, 2020-06-17:
I am sorry to re-activate this issue but I am seeing the exact same problems like @lucwuyts on my Sonoff-Tasmota 6.6.0 (which from what I can tell is way newer than 2017).

> 09:45:45 WIF: Checking connection...
> 09:45:45 WIF: Connected
> 09:45:47 MQT: Attempting connection...
> 09:45:52 MQT: Connect failed to 192.168.68.112:1883, rc -2. Retry in 10 sec

After reboot the system will work well for some hours and then continuously fail with these errors.
What does the return code rc -2 exactly mean?

Any ideas how to solve this?
Thanks!
Best regards
Thomas