arendst/Tasmota #80: Default MQTT_TOPIC as unique

Nayar, 2017-02-18:
It would be great if the default MQTT_TOPIC would be unique just like MQTT_CLIENT_ID. If you're flashing in bulk, it becomes a pain to setup different MQTT_TOPICs for each Sonoff.

I got a feeling that when I set the MQTT_TOPIC the same as MQTT_CLIENT_ID, the Sonoff seems to behave badly after sometimes. 

Another way might be to use the client_id as publish topic if the topic id is `sonoff`.

lucwuyts, 2017-02-18:
I believe you can query all topics, and change them with a mqtt command.
(see:  https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)
So this is a scriptable task.

First query the grouptopic with Status=5 
Then you can loop through the answers to change the topic.

davidelang, 2017-02-19:
There already responds to a unique topic, the problem with setting any unique topic for sending is the question of what to use. Basing it on mac address isn't very friendly and anything else is going to be hard to make unique.

every installation will want a different standard for naming. It's far better to do this in a script after the fact as lucwuyts suggests

lucwuyts, 2017-02-19:
If someone needs a script to do this, this is a node-red flow to do just that.   The function "Rename Topic" has to be adapted with your naming rules.

`[{"id":"42ae07b5.09a278","type":"mqtt out","z":"3081621d.fb653e","name":"","topic":"","qos":"","retain":"","broker":"82d780f7.7d288","x":430,"y":80,"wires":[]},{"id":"2b6ffef1.c13a52","type":"inject","z":"3081621d.fb653e","name":"Get Status van ALLE sonoffs","topic":"cmnd/sonoffs/status","payload":"5","payloadType":"str","repeat":"300","crontab":"","once":true,"x":210,"y":80,"wires":[["42ae07b5.09a278"]]},{"id":"48c3e37f.e0d16c","type":"mqtt in","z":"3081621d.fb653e","name":"","topic":"stat/+/STATUS5","qos":"2","broker":"82d780f7.7d288","x":140,"y":140,"wires":[["8d3f403e.bc24b"]]},{"id":"9f8a38ba.a3d228","type":"function","z":"3081621d.fb653e","name":"Save in variable  \"sonoffs\"","func":"// flow variable : sonoffs -> array {hostname, ip}\n// \nvar status = msg.payload.StatusNET;\n\n\nvar sonoffs= flow.get(\"sonoffs\");\nif( sonoffs === undefined )\n{\n    sonoffs = [];\n}\n\n// hostname = mqtt_topic - laatste 4 karaketers mac (decimaal)\nvar hostname = status.Host.substr(0,status.Host.length - 5);\n\n// zoek of host al bestaat in deze array\nvar idx = -1;\nfor(i=0; i<sonoffs.length ; i++ )\n{\n    if( sonoffs[i][0] == hostname )\n    {\n        idx = i;\n    }\n}\nif( idx == -1 )\n{\n    sonoffs.push( [] );\n    idx = sonoffs.length - 1;\n}\nsonoffs[idx] = [hostname , status.IP ];\nflow.set( \"sonoffs\" , sonoffs );\nreturn msg;\n","outputs":1,"noerr":0,"x":570,"y":140,"wires":[[]]},{"id":"8d3f403e.bc24b","type":"json","z":"3081621d.fb653e","name":"","x":330,"y":140,"wires":[["9f8a38ba.a3d228"]]},{"id":"d57f1675.e43488","type":"comment","z":"3081621d.fb653e","name":"Get all SonOff devices","info":"","x":120,"y":40,"wires":[]},{"id":"f9514665.fbb058","type":"inject","z":"3081621d.fb653e","name":"Klik om te starten","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":160,"y":240,"wires":[["36bffb4e.69a8e4"]]},{"id":"27884704.952ce8","type":"comment","z":"3081621d.fb653e","name":"Alle Topics hernoemen","info":"!! naamgeving wordt in de functie bepaald !!\nroutine is als test bedoeld.\n","x":120,"y":200,"wires":[]},{"id":"36bffb4e.69a8e4","type":"function","z":"3081621d.fb653e","name":"Rename Topic","func":"var sonoffs = flow.get(\"sonoffs\");\n\n// rename the topic\nfor(i=0; i<sonoffs.length ; i++ )\n{\n    var dev = sonoffs[i];\n    msg.topic = \"cmnd/\"+dev[0]+\"/Topic\";\n    msg.payload = \"newTopic\"+i;\n}\n\nreturn msg;","outputs":1,"noerr":0,"x":380,"y":240,"wires":[["bc6a8aa9.761718"]]},{"id":"bc6a8aa9.761718","type":"mqtt out","z":"3081621d.fb653e","name":"","topic":"","qos":"","retain":"","broker":"82d780f7.7d288","x":610,"y":240,"wires":[]},{"id":"82d780f7.7d288","type":"mqtt-broker","z":"","broker":"HOME","port":"1883","clientid":"","usetls":false,"verifyservercert":true,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willRetain":null,"willPayload":"","birthTopic":"","birthQos":"2","birthRetain":"true","birthPayload":""}]`

Nayar, 2017-02-20:
Did it with my mosca mqtt server:

```
server.on('published', function(packet, client) {
    publish = {
        packet: packet,
        client: client
    }
    try{ 
        packet.payload = new String(packet.payload)
        packet.payload = JSON.parse(packet.payload)
        
        if(packet.topic.search(/\/sonoff\//) > -1){
            var vals = client.id.split("DVES_");
            var packet = {
              topic: 'cmnd/' + client.id + '/topic',
              payload: 'SONOFF_' + vals[1],
              qos: 1,
              retain: false,  
            };

            console.log('MQTT broker sending message to client ...');

            server.publish(packet, function() {
              console.log('Topic set');  // it passes by here
            });
        }
        
        console.log(JSON.stringify(packet));
    } catch (err) {
        console.log(JSON.stringify(packet));
    }
});
```

If the topic matches the default one, it will send another command to rewrite it

closed 2017-02-20 by Nayar